openhardwaremonitor/openhardwaremonitor #869: CPU temp not correct show: |  +- Core #1 - #4   :        0        0        0 (/amdcpu/0/temperature/0)

a421809265, 2016-10-31:
here is the report by the software:

Open Hardware Monitor Report

--------------------------------------------------------------------------------

Version: 0.7.1.0

--------------------------------------------------------------------------------

Common Language Runtime: 4.0.30319.17929
Operating System: Microsoft Windows NT 6.1.7600.0
Process Type: 64-Bit

--------------------------------------------------------------------------------

Sensors

|
+- Gigabyte GA-880G-UD3H (/mainboard)
|  |
|  +- ITE IT8718F (/lpc/it8718f)
|  |  +- CPU VCore      :    1.344    0.992    1.344 (/lpc/it8718f/voltage/0)
|  |  +- DRAM           :    1.584    1.584    1.584 (/lpc/it8718f/voltage/1)
|  |  +- +3.3V          :    3.328    3.328    3.344 (/lpc/it8718f/voltage/2)
|  |  +- +5V            :  5.16096  5.13408  5.16096 (/lpc/it8718f/voltage/3)
|  |  +- Voltage #5     :    3.008    3.008     3.04 (/lpc/it8718f/voltage/4)
|  |  +- Voltage #6     :     4.08     4.08     4.08 (/lpc/it8718f/voltage/5)
|  |  +- Voltage #7     :     4.08     4.08     4.08 (/lpc/it8718f/voltage/6)
|  |  +- Voltage #8     :    2.832    1.936    2.832 (/lpc/it8718f/voltage/7)
|  |  +- VBat           :    3.104    3.104    3.104 (/lpc/it8718f/voltage/8)
|  |  +- Temperature #1 :       41       41       41 (/lpc/it8718f/temperature/0)
|  |  +- Temperature #2 :       51       44       52 (/lpc/it8718f/temperature/1)
|  |  +- Temperature #3 :       47       47       47 (/lpc/it8718f/temperature/2)
|  |  +- Fan #1         :     1875  1804.81  1923.08 (/lpc/it8718f/fan/0)
|  |  +- Fan Control #1 :                            (/lpc/it8718f/control/0)
|  |  +- Fan Control #2 :                            (/lpc/it8718f/control/1)
|  |  +- Fan Control #3 :      100      100      100 (/lpc/it8718f/control/2)
|
+- AMD Phenom II X4 B40 Processor (/amdcpu/0)
|  +- Bus Speed      :  200.912  200.912  200.917 (/amdcpu/0/clock/0)
|  +- CPU Core #1    :  3013.68  803.646  3013.76 (/amdcpu/0/clock/1)
|  +- CPU Core #2    :  3013.68  803.646  3013.76 (/amdcpu/0/clock/2)
|  +- CPU Core #3    :  3013.68  803.646  3013.76 (/amdcpu/0/clock/3)
|  +- CPU Core #4    :  3013.68  803.646  3013.76 (/amdcpu/0/clock/4)
|  +- Core #1 - #4   :        0        0        0 (/amdcpu/0/temperature/0)
|  +- CPU Total      :  37.6923  7.35295    56.25 (/amdcpu/0/load/0)
|  +- CPU Core #1    :       40  14.7059      100 (/amdcpu/0/load/1)
|  +- CPU Core #2    :  55.3846  2.94118       75 (/amdcpu/0/load/2)
|  +- CPU Core #3    :  29.2308        0  52.2388 (/amdcpu/0/load/3)
|  +- CPU Core #4    :  26.1538  11.7647  51.4925 (/amdcpu/0/load/4)
|
+- Generic Memory (/ram)
|  +- Memory         :  85.3222  85.2538  85.9248 (/ram/load/0)
|  +- Used Memory    :  4.68985  4.68609  4.72297 (/ram/data/0)
|  +- Available Memory : 0.806786 0.773663 0.810547 (/ram/data/1)
|
+- ATI FirePro V3700 (FireGL) (/atigpu/0)
|  +- GPU Core       :     1.25     1.25     1.25 (/atigpu/0/voltage/0)
|  +- GPU Core       :      110      110      110 (/atigpu/0/clock/0)
|  +- GPU Memory     :      950      950      950 (/atigpu/0/clock/1)
|  +- GPU Core       :       42       41       42 (/atigpu/0/temperature/0)
|  +- GPU Core       :        0        0        2 (/atigpu/0/load/0)
|  +- GPU Fan        :       24       24       24 (/atigpu/0/control/0)
|
+- ATI Radeon HD 4250 (/atigpu/2)
|  +- GPU Core       :   562.06   562.06   562.06 (/atigpu/2/clock/0)
|  +- GPU Memory     :      667      667      667 (/atigpu/2/clock/1)
|  +- GPU Core       :      100      100      100 (/atigpu/2/load/0)
|
[... 561 lines cut ...]
 C4 Reallocation Event Count           000000000000 200   200   0     -       
 C5 Current Pending Sector Count       000000000000 200   200   0     -       
 C6 Uncorrectable Sector Count         000000000000 200   200   0     -       
 C7 UltraDMA CRC Error Count           000000000000 200   200   0     -       
 C8 Write Error Rate                   000000000000 200   200   0     -       

Logical drive name: C:\
Format: NTFS
Total size: 32111587328
Total free space: 7642685440

Logical drive name: D:\
Format: NTFS
Total size: 107380994048
Total free space: 60310634496

Logical drive name: E:\
Format: NTFS
Total size: 107380994048
Total free space: 51904925696

Logical drive name: F:\
Format: NTFS
Total size: 73081577472
Total free space: 42695933952

DarthDestroyer, 2016-10-31:
You should try the latest Version: http://openhardwaremonitor.org/files/openhardwaremonitor-v0.7.1.5-alpha.zip

akej74, 2016-11-07:
New version is here, **0.8.0 Beta**, please try that as well: http://openhardwaremonitor.org/

a421809265, 2016-11-21:
thanks! maybe the reason becauses that i change and open the cpu core from
3 to 4.
i will try your idea.

mcnuby, 2018-06-27:
I have the same problem where my cpu temp is at 0 but the fans work fine. Everything worked fine until i activated my 4 cores and now in openhardwaremonitor my cpu temps are at 0, everything else is working fine.

DarthDestroyer, 2018-06-28:
@mcnuby could you try: http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

and attach a report here? That would help ;-)

mcnuby, 2018-06-29:
doesnt matter which version i get, ive tried all the versions

mcnuby, 2018-06-29:
Is there a way to reset or make openhardwaremonitor re register my hardware