openhardwaremonitor/openhardwaremonitor #871: Total LBA written incorrect

skibbipl, 2016-11-07:
I have Samsung 830 Pro SSD drive as a system drive. I've updated to latest OHM 0.8 and I can see new Total LBA Written sensor. It shows 322,8 GB, but its not correct. Samsung Magician shows 16.30 TB written to the disk so OHM definitely displays wrong value.
Is there anything I can do to help you debug and fix this issue?

moellerm, 2016-11-08:
Please create a report file with the OHM (File -> Save Report...) and attach it here.
And also note the exact value from Samsung Magician at the same time.

skibbipl, 2016-11-08:
OK, I think I know what's the issue. My samsung SSD device reports total writs like this (from OHM report):
`F1 Total LBA Written                  F52091290800 99    99    0     332.5353`
But the value is called total LBA not total bytes. My disk has following bytes per sector:
`Bytes Per Sector  :                512`
Basically total LBA != Total bytes written. I suppose OHM should use following formula to display written data:
` Total bytes written = Total LBA written * Bytes Per Sector`
Displaying LBA writes is misleading.

azachert, 2016-11-09:
I will try to look at this tonight, as I was the last one messing with this code,

moellerm, 2016-11-09:
I have it fixed almost, will commit it in a few minutes.

moellerm, 2016-11-09:
I have tried to fix the issue in commit 336a4b5a0cb94dd5422d1815076ed72ff044c97d

You can find a new alpha version here: http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.1-alpha.zip

skibbipl, 2016-11-09:
Looks OK. Thanks for improving this feature.
[image]

moellerm, 2016-11-10:
Thx for testing

closed 2016-11-10 by moellerm