openhardwaremonitor/openhardwaremonitor #872: Nuvoton NCT6791D - Incorrect VCore

488GTB, 2016-11-08:
On my Asrock Z170 Extreme4 with Nuvoton NCT6791D, VCore is being reported incorrectly at half its actual voltage.

Both HWMonitor and HWiNFO report the correct VCore voltage.

Could you please fix this?

moellerm, 2016-11-08:
Please create a report file with the OHM (File -> Save Report...) and attach it here.

488GTB, 2016-11-10:
Hi Michael,

Here is the report you requested.

Regards

Open Hardware Monitor Report

--------------------------------------------------------------------------------

Version: 0.8.0.0

--------------------------------------------------------------------------------

Common Language Runtime: 4.0.30319.42000
Operating System: Microsoft Windows NT 10.0.14393.0
Process Type: 64-Bit

--------------------------------------------------------------------------------

Sensors

|
+- ASRock Z170 Extreme4 (/mainboard)
|  |
|  +- Nuvoton NCT6791D (/lpc/nct6791d)
|  |  +- CPU VCore      :    0.376    0.368    0.648 (/lpc/nct6791d/voltage/0)
|  |  +- Voltage #2     :    1.712    1.704    1.712 (/lpc/nct6791d/voltage/1)
|  |  +- AVCC           :    3.392    3.376    3.392 (/lpc/nct6791d/voltage/2)
|  |  +- 3VCC           :    3.392    3.376    3.392 (/lpc/nct6791d/voltage/3)
|  |  +- Voltage #5     :    1.024    1.016    1.024 (/lpc/nct6791d/voltage/4)
|  |  +- Voltage #6     :     1.12    1.112     1.12 (/lpc/nct6791d/voltage/5)
|  |  +- Voltage #7     :     1.12     1.12     1.12 (/lpc/nct6791d/voltage/6)
|  |  +- 3VSB           :    3.344    3.344     3.36 (/lpc/nct6791d/voltage/7)
|  |  +- VTT            :     1.12    1.112     1.12 (/lpc/nct6791d/voltage/9)
|  |  +- Voltage #11    :    0.008    0.008    0.008 (/lpc/nct6791d/voltage/10)
|  |  +- Voltage #12    :      1.2      1.2    1.208 (/lpc/nct6791d/voltage/11)
|  |  +- DRAM           :    1.352    1.352    1.352 (/lpc/nct6791d/voltage/12)
|  |  +- Voltage #14    :     1.28     1.28     1.28 (/lpc/nct6791d/voltage/13)
|  |  +- Voltage #15    :    1.208    1.208    1.208 (/lpc/nct6791d/voltage/14)
|  |  +- CPU Core       :       32     30.5       60 (/lpc/nct6791d/temperature/0)
|  |  +- Temperature #1 :       35       32       37 (/lpc/nct6791d/temperature/1)
|  |  +- Temperature #2 :       30       29       30 (/lpc/nct6791d/temperature/2)
|  |  +- Temperature #3 :       19       19       20 (/lpc/nct6791d/temperature/3)
|  |  +- Temperature #5 :       15       15       15 (/lpc/nct6791d/temperature/5)
|  |  +- Temperature #6 :       15       15       15 (/lpc/nct6791d/temperature/6)
|  |  +- CPU Fan        :      441      433     1214 (/lpc/nct6791d/fan/1)
|  |  +- Chassis Fan 0  :      515      497      515 (/lpc/nct6791d/fan/3)
|  |  +- Chassis Fan 1  :      478      472      479 (/lpc/nct6791d/fan/4)
|  |  +- Fan Control #1 :  29.0196  29.0196  29.0196 (/lpc/nct6791d/control/0)
|  |  +- Fan Control #2 :  27.8431  24.7059  66.2745 (/lpc/nct6791d/control/1)
|  |  +- Fan Control #3 :  29.0196  29.0196  29.0196 (/lpc/nct6791d/control/2)
|  |  +- Fan Control #4 :  28.2353  28.2353  28.2353 (/lpc/nct6791d/control/3)
|  |  +- Fan Control #5 :  28.2353  28.2353  28.2353 (/lpc/nct6791d/control/4)
|  |  +- Fan Control #6 :  64.7059  64.7059  64.7059 (/lpc/nct6791d/control/5)
|
+- Intel Core i7-6700K (/intelcpu/0)
|  +- Bus Speed      :    100.2    100.2    100.2 (/intelcpu/0/clock/0)
|  +- CPU Core #1    :   1402.8    801.6  4408.82 (/intelcpu/0/clock/1)
|  +- CPU Core #2    :   1402.8    801.6  4408.82 (/intelcpu/0/clock/2)
|  +- CPU Core #3    :   1402.8    801.6  4408.82 (/intelcpu/0/clock/3)
|  +- CPU Core #4    :   1402.8    801.6  4408.82 (/intelcpu/0/clock/4)
|  +- CPU Core #1    :       27       25       60 (/intelcpu/0/temperature/0)
|  +- CPU Core #2    :       29       27       62 (/intelcpu/0/temperature/1)
|  +- CPU Core #3    :       28       27       60 (/intelcpu/0/temperature/2)
|  +- CPU Core #4    :       27       25       59 (/intelcpu/0/temperature/3)
|  +- CPU Package    :       32       31       62 (/intelcpu/0/temperature/4)
|  +- CPU Total      :  7.22976  3.51563  99.6094 (/intelcpu/0/load/0)
|  +- CPU Core #1    :  9.36298  2.34375      100 (/intelcpu/0/load/1)
|  +- CPU Core #2    :   6.2748        0      100 (/intelcpu/0/load/2)
[... 936 lines cut ...]
 07 Seek Error Rate                    DEDA69000000 60    68    30    -       
 09 Power-On Hours (POH)               4A1F00000000 91    91    0     8010    
 0A Spin Retry Count                   000000000000 100   100   97    -       
 0C Power Cycle Count                  5D0E00000000 97    97    20    3677    
 B7 SATA Downshift Error Count         010000000000 99    99    0     1       
 B8 End-to-End error                   000000000000 100   100   99    -       
 BB Reported Uncorrectable Errors      000000000000 100   100   0     0       
 BC Command Timeout                    010000000000 99    100   0     1       
 BD High Fly Writes                    010000000000 99    99    0     -       
 BE Temperature Difference from 100    1D00191D0000 61    71    45    29      
 BF G-sense Error Rate                 000000000000 100   100   0     -       
 C0 Emergency Retract Cycle Count      300000000000 100   100   0     -       
 C1 Load Cycle Count                   BD2300000000 96    96    0     -       
 C2 Temperature                        1D0000000D00 40    29    0     29      
 C5 Current Pending Sector Count       000000000000 100   100   0     -       
 C6 Uncorrectable Sector Count         000000000000 100   100   0     -       
 C7 UltraDMA CRC Error Count           000000000000 200   200   0     -       
 F0 Head Flying Hours                  051F0000688B 253   100   0     -       
 F1 Total LBAs Written                 C12913380200 253   100   0     -       
 F2 Total LBAs Read                    7A359D960400 253   100   0     -       

Logical drive name: E:\
Format: NTFS
Total size: 2000263573504
Total free space: 770000576512

RdWing, 2018-02-17:
I'm also having this issue. Same board, same symptoms.
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1733523/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2018-02-17:
The Issue is over one year old and you dont use the latest Version (same as the guy one year before!)

link to latest Version: http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

RdWing, 2018-02-18:
The same issue exists with the 0.8.0.3 you posted.

DarthDestroyer, 2018-02-18:
Could you attach a Report?

RoelGo, 2018-02-25:
Hi I have the same configuration and the same problem:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1756026/OpenHardwareMonitor.Report.txt)

I assume we could fix it wit the right parameters?
Any idea where to look for these?

DarthDestroyer, 2018-02-25:
Could you check it with #884 ?

RoelGo, 2018-02-25:
I just had a look and the pull request didn't solve the issue. 
But after comparing with HWMonitor I noticed that setting the Ri parameter to 1 seems to give accurate results for the Vcore.