openhardwaremonitor/openhardwaremonitor #873: Missing Support: B150 Chipset Motherboard

StKiel, 2016-11-08:
E.g. ASUS B150 series.

The Name is detected properly, e.g. ASUS B150 Pro Gaming, but:

The program does not show any info from the sensors of the MoBo: Vans, Temp, Voltage etcetera...

DarthDestroyer, 2016-11-08:
Please create a report file with the OHM (File -> Save Report...) and attach it here.

StKiel, 2016-11-09:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/580300/OpenHardwareMonitor.Report.txt)

StKiel, 2016-11-09:
The i5 CPU (here a i5 6600K) is obviously not properly supported too.
In idle (with a black DTbackground) it displays 3804 MHz, 3904 MHz, rarely 37/3604 MHz and very seldom 3504 MHz. It never display numerals below.
The displayed Min for each core is 801 MHz and Max is 3905 MHz.