openhardwaremonitor/openhardwaremonitor #877: Trojan: Win32/Pellate.A!cl detected in 0.8.0 beta

vorotech, 2016-11-22:
Windows Defender has detected Trojan: Win32/Pellate.A!cl when downloading the 
http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0-beta.zip

DarthDestroyer, 2016-11-22:
#845 #843

It is a false warning!

moellerm, 2016-11-24:
I have checked the file hosted on the http://openhardwaremonitor.org/ site once more. No trojan could be identified by the following tools

https://www.virustotal.com/en/file/eba0f937f37901d86a245a4963049462c8ae30beafb05f08afed1bcdcc13634d/analysis/1479970524/

https://www.metadefender.com/#!/results/file/YTE2MTEyNFMxLWxFOFdNNGZsSDFNZVY4V3pWemU/extracted/files

https://virusscan.jotti.org/en-US/filescanjob/b1q8fvtm91

I would suggest that the developers of Windows Defender check their detection engine once more. If there is any doubt, please build the Open Hardware Monitor directly from the source code available here

https://github.com/openhardwaremonitor/openhardwaremonitor

closed 2016-11-24 by moellerm

vorotech, 2016-11-28:
Thanks @moellerm