openhardwaremonitor/openhardwaremonitor #882: Missing Clocks / Temperatures for Xeon E5

mmehnert, 2016-11-30:
I have a Xeon E5 1650 v3 (Haswell-EP) installed on an ASRock X99 Extreme3 with Windows 10 running.
OpenHardware Monitor 0.8.0. beta lists the CPU cores, but for clock speeds and temperatures only displays a "-" for each core.
A different tool (CoreTemp from http://www.alcpu.com/CoreTemp/) has no problem getting the temperature values  - that's why I guess the problem must be somewhere on this side... :-)

DarthDestroyer, 2016-11-30:
It would be awesome, if you post a Report at this Issue! :)

mmehnert, 2016-12-03:
Ah :-) I did not realize that the report option is there.
I created a report and right at the beginning the following error occurred (I translated the parts that were in the native language of the Windows installation):

    Status: Installing driver "P:\Management\Software\openhardwaremonitor-v0.8.0-beta\OpenHardwareMonitor\OpenHardwareMonitorLib.sys" failed and file exists⏎
    First Exception: Service already exists⏎
    Second Exception: StartService returned the error: [translated: the system could not find the specified path] (Exception  HRESULT: 0x80070003)

Since P: is a network share I guess that openhardwaremonitor somehow switches privileges and in the process looses the credentials to the mapped drive.

When copying the program to a local drive, everything is fine.