openhardwaremonitor/openhardwaremonitor #883: After install the software the speaker continously sound alarm.

fuchura, 2016-12-01:
Hi,
How can I uninstall Open Hardware Monitor? 
I'm using windows 7.  In Control Panel  (Add and remove programs) don't appear the program.
I'll aprecciate any help.
Thanks

DarthDestroyer, 2016-12-01:
I cant understand your Issue and your Problem!

I think, you just downloaded the latest Version, extracted the Folder and started the application?
When this is true, just delete the Folder or the Files! There is nothing "installed"

fuchura, 2016-12-02:
Yes, I did exactly what you are saying. I understand what do you say about there is nothing installed. 
So I supose that there isn't any file running in background (after delete the folder and files) causing that the speaker sounds nonstop, isn't? I asked you because this began to happen inmediatly after open the program.  I can silence the speaker  setting the "pc beep" volumen to off. I think is a hardware problem, if you don't imagine something else that I could try then I will restore the software from factory.
Thx.

DarthDestroyer, 2016-12-02:
there should nothing stay in your System! 

i would love to see a report here with your device data! 
yes, you have to start it again, but we could take a look into your Problem!