openhardwaremonitor/openhardwaremonitor #885: Missing item display video card temperature

robertgood86, 2016-12-08:
Lost temperature display video card, GTX 780m. Previously, it worked fine. Open Hardware monitor version 0.8.0 Beta

[image]

robertgood86, 2016-12-08:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/639153/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2016-12-08:
Okay... 

Since it is a Notebook or other portable Device, i would bet that the Problem is there! 

At Idle, the Dedicated GPU will shit down and is not reachable in this State for OHWM... 

Could you try it, when you have decent load on the GPU? Running a Game or something else? 
Or you could disable the GPU Shutdown / Switch to onboard?

robertgood86, 2016-12-08:
I do not know what it was, but now I have tried to restart OHWM (File-> Reset), and it worked, video card temperature began to appear. But restarting (File->Reset) helps only if the Dedicated GPU is already active, if this one is not activated, and the temperature will not be displayed. The problem is that before did not have to manually restart (File->Reset) the OHWM, and the temperature of the Dedicated GPU is displayed immediately after the automatic start of the OHWM, together with Windows.

DarthDestroyer, 2016-12-08:
yeah, i understand... 
there are some Reasons, why it had worked before! maybe windows, winforms or anything else had used the gpu at startup and you never recognized this... 
Or your Dedicated GPU was active cause of other Reasons.

One thing is sure, if the dedicated GPU is sleeping, you will never get any informations from ohwm. It depends, that you get "some" data or nothing. But it is not reachable through the nvidia api, its not ohwm fault!

Skrapion, 2017-03-15:
But it *is* OHWM's fault that it won't show the data until OHWM is reset. Ideally it would detect when any new sensor appears, but at the very least, I'd like it to poll any sensors I have in set up to appear in my system tray and show them once they're available.

Skrapion, 2017-03-15:
Related #892

DarthDestroyer, 2017-03-15:
i hadnt tried a fix, yet... 

but, to say the truth, im not in the need of! 
of course, a laptop and its hardware is sensorable... but what should be done for that? 

for example: force dedi-gpu to be online (only for sensor data) will let battery users cry, 
let the user / gpu decide if the data is worth looking, the gpu should work fine (for example: running a game or anything that uses the dedi gpu) 

i dont like it, how it is now! but it depends more on ati / nvidia to allow offline dedi gpu data collecting... or atleast caching the last data... 

sorry, so far...

DarthDestroyer, 2017-03-15:
the only way, how other software allow this, is to put load on the gpu. 
the gpu api doesnt allow a wake up (yet)

Skrapion, 2017-03-15:
Yeah, I suspect the best solution is the HWiNFO approach, which appears to just put a load on the GPU at startup. If the card is awake when OHWM starts, even just momentarily, everything is fine. The card goes to sleep, and the sensor reports 0 until it wakes again.