openhardwaremonitor/openhardwaremonitor #886: Missing Support: Z10A SLI Motherboard

GingaVytis, 2016-12-09:
Name is detected properly but no other information is shown.
MSI Z170A SLI (MS-7998)
Attached report to post.
[OpenHardwareMonitor.Report(Z170A SLI).txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/641141/OpenHardwareMonitor.Report.Z170A.SLI.txt)

I'd love to help out and try and code this myself, although I'm a bit of an amateur. Is there any documentation or suggestions on how to get the motherboard sensor information?

DarthDestroyer, 2016-12-09:
You could try => #884 ;)

GingaVytis, 2016-12-09:
Awesome, got it working. Thanks so much! Next time I'll make sure to look around before I ask questions that have already been answered lul

closed 2016-12-09 by GingaVytis