openhardwaremonitor/openhardwaremonitor #889: Asus H97M-PLUS no CPU Fan Speed ?

MopheusDG, 2016-12-14:
Hi, I've been trying to get some FAN RPM speed from that motherboard with no luck. As I'm writing an app for myself using OHM I'm not sure if it's my code or the mother not being supported yet, can someone help me clarify this ?

I can read CPU Temp, Load... GPU Temp, Load... almost everything, but not that info that's in the MainBoard in particular, so my bet is not supported.

Also, it could be great to have a list of Sensors Names detailed, I saw someone already requested that here but no luck. Drive me crazy to get all info on my app.

Thanks a lot for this great OHM and hope it keeps growing.

DarthDestroyer, 2016-12-14:
When you Post a Report, it would be awesome to help you ;-)

btw, in the report are the most sensors named and self explaining. The Source is also available for everyone, when you need infos about the naming! 

When you receive no Data from the Board, your Mainboard is not supported yet and could be added by yourself or someone when we got a report!

MopheusDG, 2016-12-14:
You were right DarthDestroyer, just checked the report and helped me see the problem, not sure WHY, but in my motherboard Fans are Fan #2 (CPU) and Fan #3 (Back Fan), there's no Fan #1 at all... Sorry to have bother with this, should have read more first and get the report.

Thanks !

DarthDestroyer, 2016-12-14:
It is a little bit confusing... the numbers are set to 'first in, first out'... 
Selecting Fans staticly could do crazy things... 
atleast at the most boards, there are no names attached to the fans...

closed 2020-02-16 by MopheusDG