openhardwaremonitor/openhardwaremonitor #891: Dark theme option

JonnyRedHed, 2016-12-18:
Can we have a dark theme option please.  Dark grey background with off white text. Good for night time when the window is displaying all the time on a 2nd or 4rd screen.

Sort of like this:
https://userstyles.org/styles/37035/github-dark

Dark grey or mayne dark grey pattered BG.  Text that is light grey, which once set against the dark grey BG will be bright enough but not glaringly bright at night or in dimmed rooms.

Thank you.

JonnyRedHed, 2019-03-19:
Would be real nice to have this option. Its been years.

matdombrock, 2019-09-09:
Any chance this might ever happen?

DarthDestroyer, 2019-09-09:
Best Answer is: Maybe!

But the Sourcecode is here, downloadable and free to edit, you could make that in less then ten minutes by yourself! 

download, change the colors, compile and have fun! it doesnt cost any cent, only ten minutes...

123Inge, 2019-09-22:
This maybe a stupid question, but how do I compile this project? Really want in dark mode and want to mess around myself, but struggling to find out how to compile it. I have option to compile on both Linux and Windows. 
Thanks for any help.

DarthDestroyer, 2019-09-22:
You dont need more as Visual Studio 2019 (Community Edition) and the Source-Code (Download is on the Main-Page)

If you have both, open it up, go to the Designer and switch all Colors to whatever you like!

To Compile: Just hit compile or test it, the executable and the lib will be created and you are done!

123Inge, 2019-09-22:
Thanks a lot! Downloaded VS. Now just to find the right values!

Nowaker, 2020-06-09:
Here's a relatively easy workaround:

1. Start Open Hardware Monitor.
2. Option -> Remote Web Server -> Run
3. Install https://chrome.google.com/webstore/detail/dark-reader/eimadpbcbfnmbkopoojfekhnkhdbieeh extension in your Chrome.
4. Open http://localhost:8085/ in Chrome.
5. Set auto-refresh to 1s.

Final effect:

[image]

jesse9212, 2021-09-15:
Is there

> Here's a relatively easy workaround:
>
>     1. Start Open Hardware Monitor.
>
>     2. Option -> Remote Web Server -> Run
>
>     3. Install [...]

Is there a way to show plot (graph) on that webpage?

MRog40, 2022-10-17:
Checkout Libre Hardware Monitor. It is still maintained, and should have dark mode very soon (if still not in the release when you see this, pull my branch and build it for yourself, I'm using dark theme right now :)

Here is the PR: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/847