openhardwaremonitor/openhardwaremonitor #893: Nice, but does this help with Inspiron 1545's

AraHaan, 2016-12-22:
I have an Inspiron 1545 that can randomly shut off but I can immediately turn it back on. I am wondering if the problem could be the semi-busted USB port, the motherboard, the RAM, or a failing cooling system on it (malfunctioning fans). It does get very hot on the bottom but I am wondering if it could be anything like the heat sink or something or even the motherboard mostly. Although it does make very loud noises sometimes so I am looking at the fans as well. 

It would be nice for it to know the ranges for every part the maximum heat it handles without issue and log it so that way just right before such unexpected shutdown then we can get the actual culprit.

It could be anything that could be bad even though recently it is getting harder and harder to use anything that has a lot of graphics or even use something that used to take little to no memory now taking up a lot more than it should.

DarthDestroyer, 2016-12-22:
The Data, you are requesting is not stored into the Hardware or Software!
If i understand you correct - your _Issue_ is not really a question or an issue?!

So, back to business... 
I bet, that your Motherboard isnt supported - yet... So you lost the half of possible Sensors...
First, i would post a Report at a real Issue to request your MoBo!

After that, you will know, what you do ;)

AraHaan, 2016-12-23:
Well seems that after plugging in a USB Fan that it stops so it could be that there is a shortage somewhere in the cooling system of the laptop although sometimes it does sound like it scratches stuff so it is possibke the fans are loose and dirty or they just flat need replaced.

DarthDestroyer, 2016-12-23:
This here, is a community about ohwm as a program. 
It is not a Hardware (itself) or a support forum! 

for now, i will answer to your problem, but like i said... you are at the wrong address! 

i give you two advices... you could contact a community about your product or, you could open it up with the help of Youtube and do it yourself. 

When a Laptop doesnt work anymore and is hot, it is in the most situations a thermic problem cause of dust and dirt in the fan(s). clean it and it should solve your problem...