openhardwaremonitor/openhardwaremonitor #896: Restore tray icons after reboot

SoWhy, 2016-12-26:
OS: Win7x64

I love OHM to quickly display CPU and GPU core temp in my taskbar but after every reboot the icons are gone and I have to select them anew (and change colors again as well). Could you add an option to restore tray icons on start?

dwilhelm, 2016-12-26:
This works for me with these enabled:
Options > Start Minimized
Options > Minimize to Tray
Options > Run on Windows Startup

dwilhelm, 2016-12-26:
Since the tray info is stored in OpenHardwareMonitor.config alongside the executable, maybe you're running it from a path where that file can't be saved?

SoWhy, 2016-12-26:
Yeah, sorry, my bad. Shortly after posting this issue I noticed that v0.8.0 was up and now it works.

closed 2016-12-26 by SoWhy