openhardwaremonitor/openhardwaremonitor #901: Support for Asus H170 Pro Gaming + alarms

EmuTM, 2017-01-05:
Hello,
I would like to see support for Asus H170 Pro Gaming motherboard. 
BTW, it would be great if the software had configurable alarms for various parameters.
Thank you and best wishes with your project!

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/688472/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2017-01-05:
Already done, but you have to compile it by yourself until it got implemented!

Look here: #884 

The Alarm-Thing is another Part! ;) You could change your Issue to "only Alarms" ;)

EmuTM, 2017-01-10:
Sorry for the late reply.

I updated and compiled the code and it is working nicely. Regarding various
sensors: are the sensors enumerated always the same? For instance, I have
been able to determine that the Temperature #2 is the VRM sensor. Will VRM
sensor always be enumerated as #2?

Thank you and best regards,
Blaz

DarthDestroyer, 2017-01-10:
The short answer is: Maybe
It is a "first in, first out" Thing, that means that it could be the same but if another sensor is faster, it switches its position in counting :/ 
atleast, as far as i know!

EmuTM, 2017-01-11:
Hm.. I took a deeper look at the code and found there is also a field
called "Identifier". I hope at least this is unique, regardless of
enumeration? Otherwise it is impossible to implement temperature alarms..

Best regards,
Blaz

DarthDestroyer, 2017-01-11:
also that, maybe! 

the identifier is nothing more then "$type/$chipset/$index"... 

an alarm isnt impossible, at this point... a Temp above 100°c is always a bad idea, you know what i mean? 
maybe a global alarm if any sensor hits 90°c? for example...

closed 2018-04-06 by EmuTM