openhardwaremonitor/openhardwaremonitor #904: ASUS P8P67 EVO

KaiserZAM, 2017-01-08:
Hello, ASUS P8P67 EVO does not show the CPU temperature.
Thank you.
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/692239/OpenHardwareMonitor.Report.txt)

rmattuschka, 2017-01-09:
"Ring0

Status: _Installing driver "E:\Program Files (x86)\AnVir Task Manager\OpenHardwareMonitor\OpenHardwareMonitorLib.sys" failed_ and file exists
First Exception: CreateService returned the error: Указанная служба была отмечена для удаления. (Исключение из HRESULT: 0x80070430)
Second Exception: CreateService returned the error: Указанная служба была отмечена для удаления. (Исключение из HRESULT: 0x80070430)
"

Read above  the failed to install!
Start as Administrator or STOP your antivir.

DarthDestroyer, 2017-01-09:
correct! 

ohwm needs admin to get that data! 
the mobo data and cpu data are coming from seperate sources, so they dont depend on each other!

KaiserZAM, 2017-01-09:
Disables antivirus, run does not work as administrator.

DarthDestroyer, 2017-01-09:
why? 

You had done right-click on the executable and selected "run as admin"? 
have you said "yes" at the UAC Popup? 

If you had done all of that, please post a new report!

KaiserZAM, 2017-01-09:
Yes I have done.
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/694386/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2017-01-09:
I cant read or understrand russian, so i dont know, what it means!
for atleast i can say, that the driver couldn't be loaded!

there are some problems here and there:
1. Location: change hard-drive or directory! but i dont recommend C: (more restrictions!)
2. does your username contains any "crazy" letters? (for example russian?) - some .net apps have probs with them.
3. make 100% sure, that you had run it as admin ;)

rmattuschka, 2017-01-10:
You can read this:
http://stackoverflow.com/questions/20561990/how-to-solve-the-specified-service-has-been-marked-for-deletion-error
maybe it can you help