openhardwaremonitor/openhardwaremonitor #905: Tray Icons not saving sequence

Augusto7743, 2017-01-12:
Hello.
I have configurated to show in tray icon cpu core of 6 cores of AMD CPU FX. In orders is from left first the core 1 and after 2-3-4-5-6. After closing and restarting Open Hardware Monitor the tray icon not is same sequence 1-2-3-4-5-6 instead is 1-6-5-4-3-2. All time is need to fix the sequence moving icon cpu core 2 to second place and after cpu core 1 from second place to first place. Not have other softwares changing tray icon sequence.

Thanks very much and good luck in your path.

DarthDestroyer, 2017-01-12:
Tray Icons got managed by Windows and are not orderable or manageable through an application (like show them always) =/

Augusto7743, 2017-01-13:
Yes Windows manage tray icons, but is possible for a programa orderable a icon.
EVGA Precision X is possible choose the sequence of icons moving a icon and Windows not manage the sequence if you change the icons sequence.
Open Hardware Monitor display correctly the icons sequence if the CPU is dual core. Allways start with core frequency 1-2. CPUs six core start 1-6-5-4-3-2 and need to move core 2 and 1.
Open Hardware Monitor display icons from right to left and if just two icons are created both are displayed in sequence correctly when starting.
Is possible to add a feature to programa when starting first load each icon waiting a time to anothers icons ? If yes the issue is fixed.

smjkh, 2025-03-11:
Also having an issue where the pinned icons don't stay pinned in the taskbar when the computer wakes from sleep.  Doesn't happen for other applications either.  I keep my CPU & GPU temps in taskbar pinned and sometimes one stays pinned but often both unpin.