openhardwaremonitor/openhardwaremonitor #909: No 7700K temps

floxigen, 2017-01-17:
temps of new intel i7 7700k kaby lake cpu won't display

floxigen, 2017-01-17:
http://imgur.com/HwZhTEk

DarthDestroyer, 2017-01-17:
Please post a Report!

floxigen, 2017-01-17:
I don't really how to do this, i'm new on github

DarthDestroyer, 2017-01-17:
Open the App, Top Left => File => Save Report...
Drag & Drop it into the Textfield here and its done ;)

MrHugelberg, 2017-01-18:
I've got the same problem
I hope you don't mind that i post it here

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/714089/OpenHardwareMonitor.Report.txt)

floxigen, 2017-01-18:
I don't mind, thank you for that

moellerm, 2017-01-18:
I have uploaded an alpha version with the issue fixed:

http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.2-alpha.zip

Please test and report the results here.

MrHugelberg, 2017-01-18:
woah that was fast, tyvm!

now it works 👍 
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/714802/OpenHardwareMonitor.Report.txt)

closed 2017-01-18 by moellerm

jtugsop, 2017-01-19:
Moellerm's alpha fix works for me. I now have temps, power and clocks displaying for my 7700k.

dreness, 2017-04-01:
Consider putting a link to the alpha with this fix on the download page :)

eadmaster, 2017-05-15:
for some reason the CPU temperature reported does not match the one reported by the BIOS:
before booting Windows i've read 42°C in the BIOS, now OHM reports 30°C...
The voltages readings are also missing.
(I have a Pentium Kaby Lake G4560 with an ASRock B150M Pro4S/D3 motherboard)

DarthDestroyer, 2017-05-15:
@eadmaster, please open an Issue for your Problem and add a Report please! You are wrong here! Sorry

salighie, 2018-02-07:
Just as an FYI - I had the same issue with an i7-8700. v080-beta would only show core loads. No clocks, temps or powers... i can confirm that with v0802-alpha i get all that.

attaching reports for both versions in case it's helpful.
[OpenHardwareMonitor.Report_v080b.zip](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1704217/OpenHardwareMonitor.Report_v080b.zip)
[OpenHardwareMonitor.Report_v0802a.zip](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1704218/OpenHardwareMonitor.Report_v0802a.zip)

JMM58, 2018-07-10:
I get almost all the important info from HardWarMonitor except the Vcore readings with my i5 8600K. I use the beta 0.8.0.3.
Motherboard is the Z370 Gigabyte Aorus Ultra Gaming Wifi.
Anybody some info about this issue?

DarthDestroyer, 2018-07-10:
@JMM58 if you would add a Report here, i could lead you to the Solution! Z370 could be possible

JMM58, 2018-07-10:
Report? Dou you mean printscreen? Sry but i'm a noob with this cases and english is not my native language.

DarthDestroyer, 2018-07-10:
@JMM58, not my native language, too ;-)

You can create a Report in OHM with "File" - > Report - > Save

Something like that ;-)

JMM58, 2018-07-10:
Sry but i don't know how to that. I will wait till the crew of OpenHardwareMonitor will resolve this issue.
Thx anyway for your aid.

Whotakesmyname, 2019-05-14:
Encountered the same issue. Report attached. Thanks. @DarthDestroyer 
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/3178421/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2019-05-14:
@Whotakesmyname http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

Could you try that and ensure you are starting as admin?

JMM58, 2019-05-14:
I have for quite some time the v0.8.0.3 -alpha version but no V-core readings with my 8600K, also not as admin.

File report:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/3178584/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2019-05-14:
@JMM58

Chip ID: Unknown ITE with ID 0x8686 at 0x2E/0x2F

i think that there is a solution for your chipset, probably in the Pull Requests

Whotakesmyname, 2019-05-14:
> @Whotakesmyname http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip
> 
> Could you try that and ensure you are starting as admin?

Thanks for replying. I am sure it has been run as admin. On windows it cannot be started without admin, and if you don't manually do it, there will be a UAC message box, right? To make sure, I have also tried right click -> run as admin just now, and it appeared to be no different.

I have tried another tool, HWMonitor, and it works properly. Hope these information could help.

Whotakesmyname, 2019-05-14:
> @Whotakesmyname http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip
> 
> Could you try that and ensure you are starting as admin?

Thanks very much. The alpha version works properly for me. Hope it could be merged into the next release soon.

cnlohr, 2019-11-30:
@DarthDestroyer Even the alpha version you posted doesn't seem to work on my 9800X or any of my other Intel processors.  Any idea why processor temperature reading is so much rougher in OHW?