openhardwaremonitor/openhardwaremonitor #912: Gadget not showing NVIDIA RAM numbers.

BadlyAnticipated, 2017-01-18:
My open hardware monitor gadget is not showing the numbers of my NVidia RAM values.

It is showing the names (like: GPU Memory Free) in the gadget, but there are no values behind it.

This is the case with the following items that I want to show in my gadget:
GPU Memory Free
GPU Memory Used
GPU Memory Total

It does show the values when I look at the open hardware monitor program itself, but it does NOT show them in the gadget.

P.S. The "GPU Memory" under "Load" in the graphics card subsection, does show the name AND the progress bar, with the correct value.

Example picture:
https://i.gyazo.com/77f49b75baaa526b002bdcb1662002d5.png

DarthDestroyer, 2017-01-20:
ups :-)

thats my fault... will add a fix, when i got home! 

i added these 3 stats! 
i added a new sensortype (smalldata) for MB Values and i think, i forgot to add the display type at the gadget because i use my own gadget ;-)

you can try it by yourself, if you want ;-)

BadlyAnticipated, 2017-01-20:
If you can tell me how to do it ;) I would love to ... 

Or I'll just wait till you fixed it ;)

DarthDestroyer, 2017-01-20:
Already solved by #913 👍

samadoni, 2017-10-18:
hi,
Is this issue supposed to be fixed ?(i guess yes seeing the last post). But where can i find the compiled version with the fix?
thanks

DarthDestroyer, 2017-10-18:
Not yet implemented!

Only way to get that is compiling it yourself...

samadoni, 2017-10-18:
ok i see thanks

avengerx, 2019-03-12:
You'll only close this once the public version includes the fix? That's actually nice, we can see open issues as what's not in the "latest" version we have. Easier to find if an issue has already been reported.