openhardwaremonitor/openhardwaremonitor #914: Asus IX Hero - Missing voltages, temps, fans or fan controls, etc.

jtugsop, 2017-01-19:
First time using git here. I hope I'm posting correctly.

With an alpha fix from moellerm in another thread ohm now displays temps, power, clocks for the 7700k which is great! However there's no support for any of the motherboard temps, voltages, flow sensors, etc. Is this on anyone's radar? Thank you!

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/717816/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2017-01-20:
Yeah... Your MoBo isnt supported atm! 

=> Unknown Winbond / Nuvoton / Fintek with ID 0xD121 at 0x2E/0x2F

DarthDestroyer, 2017-01-20:
Its already added but not pulled! 

You could add it by yourself! 
Here it is =>  #884

jtugsop, 2017-01-20:
I'm afraid I have no idea how to edit the source code...

DarthDestroyer, 2017-01-21:
It isnt really hard!
Get Visual Studio 2015 Community (Community Edition is important because its free ;) )
Download the Source-Code, open it with Visual Studio and do, what is listed at the Topic i had posted.
It doesnt need much work - could be done in Minutes (okay, the installation and download would take some more time)

jtugsop, 2017-01-22:
Thank you, that's perfect. I'm confident I can edit the appropriate files and make it work. How and what I needed to accomplish this the easiest was where I was lost.

jtugsop, 2017-01-23:
I'm familiar with code, but not so much with Visual Studio. I was able to edit the code correctly, but it won't build/publish. Keeps throwing multiple ambiguous errors in the PlotPanel.cs and I'm no programmer so my troubleshooting skills are limited.

If anyone with more experience happens to include #884 into the 0.8.0.3 Alpha build it would completely round out full support for the 7700k. I plan to take another look at it this weekend when I get more time.

DarthDestroyer, 2017-01-23:
i would love to, but i cant! @moellerm is the only person, who can accept pull requests... so, you have to be patient for that :(

but with the rest, i could help you... or atleast, i would try to help ;)

jtugsop, 2017-01-24:
Ahh, okay. Good to know, thank you for being patient with a newcomer.

I'm not exactly sure what I would need help with. I applied the code changes in all the appropriate places without changing anything else and it won't publish to executable. I'm open ears for any pointers you might have, or perhaps some step I might be inadvertently missing. Thanks again.

jtugsop, 2017-03-07:
Anyone know when moellerm is returning? It looks like he hasn't been active in quite some time.