openhardwaremonitor/openhardwaremonitor #917: ASUS B150 PRO GAMING/AURA motherboard not supported

RafaelLiChen, 2017-01-21:
doesn't show sensor information like fans, voltage, etc.

DarthDestroyer, 2017-01-21:
please add a report!

File => Save Report => Upload to your Issue by drag & drop!

DarthDestroyer, 2017-01-21:
When there is this line written (in the Report)
Unknown Winbond / Nuvoton / Fintek with ID 0xD121 at 0x2E/0x2F
Here is the fix for that => #884

If there is another line / sentence or other numbers, your report would be needed!

RafaelLiChen, 2017-01-21:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/721592/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2017-01-21:
Thanks for that, your MoBo has the same Chipset which i posted above!
Im Sorry, but you have to compile that fix for yourself - atm until the fix got pulled into a new version!

RafaelLiChen, 2017-01-21:
ok, thank you