openhardwaremonitor/openhardwaremonitor #918: Temperature not shown for Intel i7-7700K

socketfailure, 2017-01-23:
Temperature not shown for Intel i7-7700K
http://i.imgur.com/PnNPuPL.png

socketfailure, 2017-01-23:
I found it in #911 

http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

i7-7700K temps now display.

mapbry, 2017-06-29:
It still won't show it for me with this alpha version.  It does not show clocks or temperature for my 7700k.

DarthDestroyer, 2017-06-29:
@mapbry, could you upload a report? 
did you start OHM as Admin?

smangelsdorf, 2017-10-07:
I was having this problem too, and have confirmed that 0.8.0.3-alpha fixes it. Thanks 👍

Froggy11, 2017-11-03:
v0.8.0.3 shows temps but wrong core frequency (800MHz instead of 4400MHz)

DarthDestroyer, 2017-11-03:
i can't confirm that!
as far as you are in "balanced" mode, put some load onto or switch to "highpower" mode and it should list your desired real frequency! 

if that doesnt work for you, a report would be nice!