openhardwaremonitor/openhardwaremonitor #919: Support request for Asus Z170-Plus

asampal, 2017-01-24:
Not showing temps, fans, voltages for mobo and voltages, utilizations for CPU (6700K). Report submitted from app. TIA for the attention.

DarthDestroyer, 2017-01-24:
You have two Options far now:

First: http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip
Second: When there is this line written (in the Report)
Unknown Winbond / Nuvoton / Fintek with ID 0xD121 at 0x2E/0x2F
Here is the fix for that => #884

It would be awesome, if you add your Report to your Issue!
File => Save Report => Upload to your Issue by drag & drop!

asampal, 2017-01-25:
Here it is.

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/728415/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2017-01-25:
Look at #884!

It is the same Chipset! 
Two Options available for that, too! 

First: Get Source, Edit it with the fix above and compile it
Second: Wait until the pull request is accepted and get the new version

Sorry for that

asampal, 2017-01-25:
It's not clear to me if you were suggesting 0.8.0.3 alpha had the fix for #884 already.

DarthDestroyer, 2017-01-25:
sorry, the latest Version doesnt include it! 
it was only for the latest cpu support!