openhardwaremonitor/openhardwaremonitor #924: OHM constantly scans HDD disks when shows CPU temperature in tray.

maxim-boligatov, 2017-01-27:
Program constantly scans HDD disks when it runs.
But it should not do that when it is minimized and shows CPU temperature in system tray.
Otherwise such behavior will not allow to power off HDD discs in notebooks (in correspondence with power plan).
So for now hard disks are periodically "on" and "off" (spin started.. stopped.. started stopped.. very annoying).

DarthDestroyer, 2017-01-27:
Disable HDD would solve your Problem!
Of course it is needed to start the Drive to get its Data... If you dont want that, switch it off ;)

maxim-boligatov, 2017-01-28:
I am not sure that I understand. What exactly should be disabled?
Hard disk? Nice joke... Why not remove it at all in this case?

Power consumption in notebooks is most important thing when you are out of "cord-zone".
Often notebooks can have 2 hard drives (even can be more). And mostly always it is not required for additional HDD to be up and running all time. So it is normal case when OS switch power off for this particular device (and starts it only if we really access files on that HDD). But OHW prevents such behaviour. He constantly access drive... so OS cannot switch it off. I think it is obvious to scan only specific parameters when program minimized. For example scan only ones checked in tree with sensors (including displayed in system tray).

DarthDestroyer, 2017-01-28:
You got me wrong!
You can "disable" the HDD check in OHWM!

File => Hardware => HDD (uncheck it)
(You can uncheck unneeded things, too - for optimal power usage)

When you need it or it is plugged in, you can enable it and check your HDDs!
To explain: If you have a Car, do you know if the engine works when you havent started it?
I wish the same thing would be possible with the GPU at Notebooks... When you have a dedicated GPU, it will only get the Data, when the GPU is in use (because of the sleep mode)

I hope, you got me now!
BTW: I use the HDD Check once in a Month - my two external HDDs use so much time to start, that OHWM has its problems and an epic loading time ;)

maxim-boligatov, 2017-01-29:
> File => Hardware => HDD (uncheck it)
I think it can be a solution. Thanks!

closed 2017-01-29 by maxim-boligatov