openhardwaremonitor/openhardwaremonitor #925: CPU Clock Speed Incorrect (i7-6700)

Pointeh, 2017-01-30:
I've had a look around, hope I'm submitting this Issue correctly.

Newly overclocked I7-6700 (none K) Core Clock is not reflected correctly in OpenHardwareMonitor.
Seems like the bclk is still assumed to be 100Mhz whereas it's nearer 133 now.

Value reported correctly in other tools, http://i.imgur.com/6QmUIdf.png

Attached the report for your perusal.
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/739614/OpenHardwareMonitor.Report.txt)

I have tested on the latest 8.0.3 Alpha with the same results.

Pointeh, 2017-01-30:
I should also point out that the missing temps expected. It is a side effect of the 'SKY OC' method to OC None-K CPUs.

floxigen, 2017-02-02:
Confirmed

http://i.imgur.com/f00QkWW.png

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/748838/OpenHardwareMonitor.Report.txt)