openhardwaremonitor/openhardwaremonitor #926: Voltages, Temperatures and Fan sensors Won't show on board MSI Z270M Mortar with NUVOTON NCT6795 and intel I7-7700K

SubFuzz, 2017-02-03:
Hello,
I'm pretty new at this, the motherboard MSI Z270 Mortar (model MS-7A69) doesn't show the voltages, Températures and Fans sensors that would show on my previoux board/cpu by Asus P8Z77-M Pro ans I5 3570 (unknown i/o chip).
Program version used is the 0.8.0 Beta and 0.8.0.3-Alpha both run as administrator, here's a capture : 
[image]

Thanks and feel free to ask any other détails.

TheBhim, 2017-02-13:
Hi, same here with MSI Z270 GAMING CARBON mainboard and Intel Core i7 7700k too. Any way to help with this? Any data you (meaning the developer, not issue opener^^) might need to identify sensors maybe?

DarthDestroyer, 2017-02-13:
#930 #884 

It would be the best, if you upload a report!
i recommend to upload it to this issue!

SubFuzz, 2017-02-14:
Here you go, thanks for the comments!

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/772724/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2017-02-14:
Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xD352 at 0x4E/0x4F

TheBhim, 2017-02-14:
Hi, CPU works fine now that I switched to 0.8.0.3 Alpha (sorry, missed that one) and I uploaded a report via the application even before posting here. :o)

DarthDestroyer, 2017-02-14:
The InApp Reports are landing in nowhere ;)
They are only accessable by @moellerm (the creator)

It would be awesome to collect the reports here to seperate the needed data! ;)

RaydenX93, 2017-02-27:
Report for MSI Z270 Gaming M3 (from issue #942 )

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/803274/OpenHardwareMonitor.Report.txt)

xcompon3nt, 2017-03-09:
This is what I'm getting with GB Z270X-G5, fans and CPU voltage would be great!
[image]

mbesset, 2017-04-08:
Hello,
I have the same problem as the others, no information on the Motherboard.
MSI Z270 Gamin Pro Carbon

Here is my report, I hope this will help you !
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/907552/OpenHardwareMonitor.Report.txt)

Thx

DarthDestroyer, 2017-04-08:
@mbesset, that Report is useless... Sorry! Could you check, if you had disabled the Mainboard?

mbesset, 2017-04-08:
A yes sorry, I disabled the motherboard because it does not show anything voico the new report with motherboard activate
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/907753/OpenHardwareMonitor.Report.txt)

fpajor, 2017-04-11:
Same issue here on MSI Z270-A PRO.
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/913035/OpenHardwareMonitor.Report.txt)

betz0r, 2017-07-26:
Same issue here with MSI Z270:
Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xD352 at 0x4E/0x4F
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1176591/OpenHardwareMonitor.Report.txt)

Can we implement that chipset ourself? Any doc/howto out there regarding this procedure in openhardwaremonitor?

DarthDestroyer, 2017-07-26:
@betz0r #884 <- that should help

TechnoNerdDad, 2020-03-21:
Same with MSI MPG X570 GAMING PLUS
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4363667/OpenHardwareMonitor.Report.txt)