openhardwaremonitor/openhardwaremonitor #930: Won't Show CPU or MoBo Temperature on Gigabyte Z170X-UD5-CF

mapbry, 2017-02-10:
I can see both my GPU's, and the CPU load, but it gives no other information on the CPU or the Motherboard.

DarthDestroyer, 2017-02-10:
it would be awesome, if you could upload your report!

mapbry, 2017-02-10:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/767225/OpenHardwareMonitor.Report.txt)

I'll attach it here, I have also uploaded it from within the software it'self.

DarthDestroyer, 2017-02-10:
Two Things...

First: http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip
This is the latest Version! #911
That should help you, with the CPU Data...

Second: Your Mobo isnt supported yet!
Chip ID: Unknown ITE with ID 0x8686 at 0x2E/0x2F 
Hopefully, someone could add it... but, feel free to do it on yourself ;)

lucascosti, 2017-03-01:
Just started using a Gigabyte Z270X Gaming K5 with the same Chip id (`0x8686`).

My CPU (7600K) is supported with 0.8.0.3.Alpha.

Attached is my report. Would love this to get the MB supported! :)

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/810049/OpenHardwareMonitor.Report.txt)

closed 2022-09-14 by mapbry