openhardwaremonitor/openhardwaremonitor #931: Can't safely remove external hard drive

sycorax, 2017-02-16:
If OpenHardwareMonitor is running I can't safely remove my external hard drives. Windows says they are in use. If I close OHM, then it lets me.

DarthDestroyer, 2017-02-16:
#924 is the nearly the same issue! hope it helps...

sycorax, 2017-02-17:
Yes! It works by disabling HDD controllers. Thanks.

Nonetheless I think OHM shouldn't hold HDD from removal.
I saw the same issue is with USB flashes.

DarthDestroyer, 2017-02-17:
The problem is, ohm needs online devices to check them! 
It is the same with Notebooks and there dedicated gpu, if it sleeps it will never put any data out! 

hope you get, what i mean... 

i dont know, if windows give us the ability to check when you want to safely remove a hdd... or let them sleep...

For now: If you want data, it has to be enabled and cant be removed :-(

sycorax, 2017-02-17:
Ok, maybe just keep an open ear if Windows provides some solution.
Thank you and keep the good work :)

closed 2017-02-17 by sycorax