openhardwaremonitor/openhardwaremonitor #939: OHM missing some fans and temp sensors for Asus Gryphon Z97

ragesaq, 2017-02-27:
The Asus Gryphon Z97 has a few extra fans and temp sensors than many of the other Asus Z97s. 

1: Currently it is missing the Assist Fan, a 3 pin header for a small fan that goes inside the motherboard IO plate.

2: While it has Temperature 1-5 that is from the Nuvoton NCT6791D chip, there is another area that HWiNFO64 calls "ASUS EC" where VCORE, VCORE (BACK), DRAM, PCH, USB3.0, PCIE1, PCIE2 as well as 3x 2pin thermal probe ports are that can be used for monitoring specific things. In my case I am monitoring my custom loop water cooling system. It would be fantastic to monitor this with OHM.

If there is any info I can provide that would help I would be glad to do so.
Thanks

ragesaq, 2017-03-03:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/816024/OpenHardwareMonitor.Report.txt)
Here's the report file

ragesaq, 2017-03-20:
Would love to hear about how I could help get the information necessary to ahve these features added to OHM. This additional sensor chip is used on many Asus boards, not just the TUF line. A few of the Maximus ones also feature the additional sensors.

ragesaq, 2017-06-02:
The Asus TUF motherboards, and probably others, have an extra TUF ICe chip that is either an additional chip or has some currently undetected extensions for additional temperature sensors on them. I've been doing a bit of reading around other applications that do similar things to OHM and how they have handled this issue but I'm drawing complete blanks.
CPUID HWMonitor doesn't pick them up but HWiNFO64 does.

ragesaq, 2017-06-02:
Ok, found something on the hwinfo forums, looks like an AMDA00 Interface Driver pulls info from the ASUS EC sensor.
https://www.hwinfo.com/forum/Thread-Asus-Embedded-EC-chip-get-bugged-T-Sensor-values

DarthDestroyer, 2017-06-05:
I had read that Thread - The Content isnt really helping there :(

HWiNFO is, as far as i know, closed source... That makes it really hard to get the Path, they went to get those Data! I use an Asus Board, too... But i had never needed those Data! Okay, i did not know that this exists ;) 
And, i dont have these Thermal Probes...

So, for now, we need to get more and better informations about that!

ragesaq, 2017-06-06:
Yeah, based on what I've read, there is an external driver that is polled to retrieve the information from the TUF ICE chip, which contains more sensors than just the t-probes.
[Imgur](http://i.imgur.com/8TEO8eN.png)
The last 3 temp sensors seen under the Asus EC section are the T-Probes built into my Asus Gryphon Z97 motherboard, they are inline water temp sensors that are a part of my custom loop. I'd like to get this into my system monitor that I put together, here is an export of my monitoring chart.
https://snapshot.raintank.io/dashboard/snapshot/wylhczT7EHOa3F2jCjvGQ70Vqfnh8VhZ

DarthDestroyer, 2017-06-06:
That doesnt sound bad, right now! 

i had read about "high cpu usage" and "dont run with original Ai Suite"... That could be a ko criteria! 

I have to look at the Ai Suite dll's... Maybe there is something to implement... 
If that is somehow accessable, it should be possible! 
But like i said, the HWiNFO way with the warning popup isnt great :-(