openhardwaremonitor/openhardwaremonitor #940: OHM missing nvidia 1070 power draw and core voltage

ragesaq, 2017-02-27:
Would be nice to get core voltage and total watts from GPU

ManiaProductions, 2018-01-07:
@ragesaq  hey any info on your post? would like to have power sensor for my 1080 too...

DarthDestroyer, 2018-01-07:
#787

Sad is, that i couldnt got it complete... still stuck at pstate :-(

ManiaProductions, 2018-01-07:
ah, too bad. i wish i knew more to help you

DarthDestroyer, 2018-02-08:
it isnt implemented yet, so it is not available ;-)

for the pstate, you could try my "old" fork... #787

i have to work on that, but time is a problem :-\

salighie, 2018-02-08:
I hear yah. just trying to help . wish i could contribute... i'm not a seasoned developer, but i could be dangerous.

Speaking of which, not sure if it helps, but nvidia software ships with nvidia-smi.exe (typically under C:\Program Files\NVIDIA Corporation\NVSMI) - outputs all sorts of goodies about their GPUs... maybe you can hook into its libraries, or de-compile to see how they go about it. 

I might do this myself one day, maybe compile my own fork, but as you aptly put it... time is a problem.

anyway, great work - nothing like this out there. Sure there is CPUID's Hardware monitor, but they don't expose any apis (not for free anyway) or write to WMI. 

Thanks

DarthDestroyer, 2018-02-08:
im not the founder or owner... only a contributor ;-)

feel free to try my fork! 
there isnt a big Problem to get the data, the way is already clear. But it is time consuming, which i dont have right now :-(

If you get the pstate, you can look for voltage corresponding to its pstate...

angelodemon2017, 2018-05-17:
Good afternoon, it turned out to relieve voltage?