openhardwaremonitor/openhardwaremonitor #943: No temp value on Baytrail E3845

mrsinger, 2017-03-02:
Subject says it all.   Using the .dll from 0.8.0 attempting to read the temp on the cores of E3845.  Tested my code on a J1900 and all is well.

DarthDestroyer, 2017-03-02:
when you upload a report, we could help you maybe ;-)

mrsinger, 2017-03-02:
Sorry, first timer with github.  How do I do that?

DarthDestroyer, 2017-03-02:
it isnt hard ;-) ohwm - > file - > report 
it is a textfile you can easily drag amd drop into your comment!

mrsinger, 2017-03-02:
Sorry, I thought you were referencing a github function, not the program.  I hadn't run it on the atom motherboard as my calling into the DLL worked fine on the celeron.

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/814466/OpenHardwareMonitor.Report.txt)

So the temp shows up there, but if I run this code..  I get no values on the atom, but do on a celeron.
thisComputer = new Computer() {CPUEnabled = true };
thisComputer.Open();

 foreach (var hardwareItem in thisComputer.Hardware)
.
.
.
                {
                    if (hardwareItem.HardwareType == HardwareType.CPU)
                    {
                        hardwareItem.Update();
                        foreach (IHardware subHardware in hardwareItem.SubHardware)
                            subHardware.Update();

                        foreach (var sensor in hardwareItem.Sensors)
                        {
                            if (sensor.SensorType == SensorType.Temperature)
                            {

                                temp += String.Format("{0} T = {1}\r\n", sensor.Name, sensor.Value.HasValue ? sensor.Value.Value.ToString() : "no value");

                            }
                        }
                    }
                }