openhardwaremonitor/openhardwaremonitor #949: Can't get ATI Radeon HD 4800 temps and fans by wmi

mykolq, 2017-03-09:
Can't get ATI Radeon HD 4800 temps and fans by wmi (tried powershell and zabbix_get), but in gui this data exists.
[image]
[sensors.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/830626/sensors.txt)

mykolq, 2017-03-14:
Radeon 7800 doesn't show data anyway. in gui and by wmi - the same result

mykolq, 2017-03-14:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/841535/OpenHardwareMonitor.Report.txt)

mykolq, 2017-04-05:
No news? Maybe you need more info?

mykolq, 2017-04-24:
Is there any way to fix it manually? Some steps that i have and can make ny myself?

DarthDestroyer, 2017-04-24:
You should clean that Issue, before ;)

Let me help (as far as i understood, what you said)

Values  | ATI Radeon HD 4800 | ATI Radeon HD 7800
------------ | ------------- | -------------
GUI | X | -
WMI | - | -

But you posted only one Report (HD 7800)?

That, what i see, seems to be a Radeon Driver Problem maybe? Atleast for the 7800!
If you could post a Report for your 4800, that would be awesome! I dont have a Radeon here, so your Report is your best help to investigate!

mykolq, 2017-04-25:
Radeon 4800 - no wmi, but gui ok. 7800 - nothing
[wmi.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/954116/wmi.txt)
[AMD4800.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/954117/AMD4800.txt)

mykolq, 2017-05-24:
Anybody here? Maybe i can add this vga by myself? If i can, how?