openhardwaremonitor/openhardwaremonitor #953: CPU & GPU temperature absent on Gen.7 mobile and 1000 series GPU

m3j3, 2017-03-19:
Hi there, 

I've been using OpenHardwareMonitor for a while now and it's been working great. That is, until I changed laptops. I now have a Dell Inspiron 7567 running an i7 7700HQ and GTX 1050Ti. The only temperatures showing up are SSD/HDD. 

Are you aware of this issue and do you know if there is a new version of OpenHardwareMonitor coming soon? 
[image]

DarthDestroyer, 2017-03-19:
Without a Report, it is nearly not possible to help you!

First: Your Mainboard is not supported
Second: Start the Application as Admin
Third: use the latest Version is http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

For Further assistance, we need the Report! ;)

m3j3, 2017-03-19:
Thank you for replying.
My apologies, I forgot to send the report! 
The alpha works for CPU temperatures, not getting a reading on GPU. 

Here is the report for the beta:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/853469/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2017-03-19:
The gpu is maybe sleeping, could you put some load on it and then start ohm? 

or, if you know what we do: . #952

m3j3, 2017-03-19:
Stressing GPU reveals temperatures and frequencies of GPU; it was, and you were right, sleeping. 
Here are the reports for beta and alpha:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/853497/OpenHardwareMonitor.Report.txt)
[OpenHardwareMonitor.Report[ALPHA].txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/853498/OpenHardwareMonitor.Report.ALPHA.txt)

DarthDestroyer, 2017-03-20:
I copy some Data for a support Request, first:

Chip ID: Unknown ITE with ID 0xFE00 at 0x2E/0x2F

Chip ID: Unknown SMSC with ID 0xFE00 at 0x2E/0x2F

For your GPU Problem there is a Pull Request (new Functions that could go into a new Version) open! 
If you can code, you can reproduce that pull request on your own... if not, you have to wait until it gets accepted or another solution is found! 

hope it helps you!

m3j3, 2017-03-20:
I'll wait for new updates to roll out, thanks for your help. 
In the meantime I'll stick with v0.8.0.3 Alpha, it works great.