openhardwaremonitor/openhardwaremonitor #954: CPU Load section not list all threads

Olakeen, 2017-03-20:
On Intel Hyper-Threading CPU, OHM show only physical threads instead of physical + logical.
When one thread is at 100% load, OHM show 50% load on the physical core.

If one process is stuck on one thread, I can't detect it with OHM :(

Tested on i3 and i7 processor.

[image]

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/855182/OpenHardwareMonitor.Report.txt)

martynsmall, 2017-08-24:
I was intrigued by this as well, and investigating it further have found that it's intentional behaviour - it actually gets values for each thread and then averages them per core.  I don't know why it's been designed like this because, as you say, it hides important information.

I modified my copy to expose the per-thread values, but I made a couple of assumptions that I wouldn't trust to work in all cases.  It's a change in functionality too so I doubt it would ever get merged even if it were to be implemented properly.  No doubt some people rely on the current behaviour!

I guess it could be made available as a setting or something, but nothing else works like that in the application so it doesn't really fit the style.  I also haven't been through anyway near enough of the code to know how to make bigger changes in it.

DarthDestroyer, 2017-08-24:
There are many ways, to get the core usage!

Im not deep enough into that, tried to find it out - but the .net PerformanceCounter-Way was easier and shorter (for one of my applications).
Maybe the way through the Ring0 is the limiting factor here?

I would love to see your way / idea at that!

martynsmall, 2017-08-24:
There's no issue with actually fetching the raw values, it's the way in which these are processed that needs altering.  The area I changed is in [CPULoad.cs](https://github.com/openhardwaremonitor/openhardwaremonitor/blob/e93e77d527649079ba41945510052fbd357e1e1d/Hardware/CPU/CPULoad.cs#L117), this loops through each core then subloops through each thread, obtaining the average value at line 130.

martynsmall, 2017-10-07:
I've pushed the change I made to my fork [here](https://github.com/martynsmall/openhardwaremonitor/commit/4a409fb4f7be0f6b3af9af6ee4368a4f27440dce).  As above I don't see that it's worth creating a PR for this as it's a change in behaviour.