openhardwaremonitor/openhardwaremonitor #955: Voltages section missing

tds4u, 2017-03-20:
Downloaded latest version, compiled with VS 2015 and added some ITE chips.
Now the report and web interface are showing correct voltages for my motherboard. The GUI itself don't display these nodes, although in TypeNode.cs (line 25) is triggered by debugger.

Who can help me?

DarthDestroyer, 2017-03-20:
Could you say, what is listed as Line 25?

Also, i dont know what you changed ;-)
Could you explain your steps? 

Does the Sensors get images / types in web-version?

tds4u, 2017-03-20:
TypeNode.cs: Break point triggered = yes
```
      switch (sensorType) {
        case SensorType.Voltage: 
          this.Image = Utilities.EmbeddedResources.GetImage("voltage.png");
          this.Text = "Voltages";
          break;
```

Report:
```
Parameters

|
+- Lenovo SHARKBAY (/mainboard)
|  |
|  +- ITE IT8733F (/lpc/it8733f)
|  |  |
|  |  +- Voltage #1 (/lpc/it8733f/voltage/0)
|  |  |  +- Ri [kΩ] : 0 : 0
|  |  |  +- Rf [kΩ] : 1 : 1
|  |  |  +- Vf [V] : 0 : 0
|  |  |
|  |  +- Voltage #2 (/lpc/it8733f/voltage/1)
|  |  |  +- Ri [kΩ] : 0 : 0
|  |  |  +- Rf [kΩ] : 1 : 1
|  |  |  +- Vf [V] : 0 : 0
|  |  |
|  |  +- Voltage #3 (/lpc/it8733f/voltage/2)
|  |  |  +- Ri [kΩ] : 0 : 0
|  |  |  +- Rf [kΩ] : 1 : 1
|  |  |  +- Vf [V] : 0 : 0
|  |  |
|  |  +- Voltage #4 (/lpc/it8733f/voltage/3)
|  |  |  +- Ri [kΩ] : 0 : 0
|  |  |  +- Rf [kΩ] : 1 : 1
|  |  |  +- Vf [V] : 0 : 0
|  |  |
|  |  +- Voltage #5 (/lpc/it8733f/voltage/4)
|  |  |  +- Ri [kΩ] : 0 : 0
|  |  |  +- Rf [kΩ] : 1 : 1
|  |  |  +- Vf [V] : 0 : 0
|  |  |
|  |  +- Voltage #6 (/lpc/it8733f/voltage/5)
|  |  |  +- Ri [kΩ] : 0 : 0
|  |  |  +- Rf [kΩ] : 1 : 1
|  |  |  +- Vf [V] : 0 : 0
|  |  |
|  |  +- Voltage #7 (/lpc/it8733f/voltage/6)
|  |  |  +- Ri [kΩ] : 0 : 0
|  |  |  +- Rf [kΩ] : 1 : 1
|  |  |  +- Vf [V] : 0 : 0
|  |  |
|  |  +- Voltage #8 (/lpc/it8733f/voltage/7)
|  |  |  +- Ri [kΩ] : 0 : 0
|  |  |  +- Rf [kΩ] : 1 : 1
|  |  |  +- Vf [V] : 0 : 0
|  |  |
|  |  +- Voltage #9 (/lpc/it8733f/voltage/8)
|  |  |  +- Ri [kΩ] : 0 : 0
|  |  |  +- Rf [kΩ] : 1 : 1
|  |  |  +- Vf [V] : 0 : 0
|  |  |
|  |  +- Temperature #1 (/lpc/it8733f/temperature/0)
|  |  |  +- Offset [°C] : 0 : 0
|  |  |
|  |  +- Temperature #3 (/lpc/it8733f/temperature/2)
|  |  |  +- Offset [°C] : 0 : 0
```
Web interface:
[image]

GUI:
[image]

DarthDestroyer, 2017-03-20:
Maybe an error related the voltage.png embedded resource? 

i had to look, how the web gets the images, but it looks like an error at there!
the gui has to use it through typenode, that is the difference...

tds4u, 2017-03-20:
The problem itself is that no voltage node with its children is shown. Can't figure out the problem for now.
But you're right, it seems to be an error inside the GUI application.

DarthDestroyer, 2017-03-20:
i think, that the voltage type doesnt work... 

you had posted the parameters, how does the other part looks?

tds4u, 2017-03-20:
```
 Hardware/LPC/Chip.cs   |  6 ++++++
 Hardware/LPC/IT87XX.cs | 10 ++++++++--
 Hardware/LPC/LPCIO.cs  |  3 +++
 3 files changed, 17 insertions(+), 2 deletions(-)

diff --git a/Hardware/LPC/Chip.cs b/Hardware/LPC/Chip.cs
index 94e75f6..df71f68 100644
--- a/Hardware/LPC/Chip.cs
+++ b/Hardware/LPC/Chip.cs
@@ -35,6 +35,9 @@ namespace OpenHardwareMonitor.Hardware.LPC {
     IT8721F = 0x8721,
     IT8726F = 0x8726,
     IT8728F = 0x8728,
+    IT8731F = 0x8731,
+    IT8732F = 0x8732,
+    IT8733F = 0x8733,
     IT8771E = 0x8771,
     IT8772E = 0x8772,
 
@@ -82,6 +85,9 @@ namespace OpenHardwareMonitor.Hardware.LPC {
         case Chip.IT8721F: return "ITE IT8721F";
         case Chip.IT8726F: return "ITE IT8726F";
         case Chip.IT8728F: return "ITE IT8728F";
+        case Chip.IT8731F: return "ITE IT8731F";
+        case Chip.IT8732F: return "ITE IT8732F";
+        case Chip.IT8733F: return "ITE IT8733F";
         case Chip.IT8771E: return "ITE IT8771E";
         case Chip.IT8772E: return "ITE IT8772E";
 
diff --git a/Hardware/LPC/IT87XX.cs b/Hardware/LPC/IT87XX.cs
index 1ff5808..65c2453 100644
--- a/Hardware/LPC/IT87XX.cs
+++ b/Hardware/LPC/IT87XX.cs
@@ -143,11 +143,14 @@ namespace OpenHardwareMonitor.Hardware.LPC {
       fans = new float?[chip == Chip.IT8705F ? 3 : 5];
       controls = new float?[3];
 
+      // IT8731F, IT8732F and IT8733F use a 9mV resultion  ???
+      if (chip == Chip.IT8731F || chip == Chip.IT8732F || chip == Chip.IT8733F) {
+        voltageGain = 0.009f;
+      } else
       // IT8620E, IT8628E, IT8721F, IT8728F and IT8772E use a 12mV resultion 
       // ADC, all others 16mV
       if (chip == Chip.IT8620E || chip == Chip.IT8628E || chip == Chip.IT8721F 
-        || chip == Chip.IT8728F || chip == Chip.IT8771E || chip == Chip.IT8772E) 
-      {
+        || chip == Chip.IT8728F || chip == Chip.IT8771E || chip == Chip.IT8772E) {
         voltageGain = 0.012f;
       } else {
         voltageGain = 0.016f;        
@@ -178,6 +181,9 @@ namespace OpenHardwareMonitor.Hardware.LPC {
         case Chip.IT8628E:
         case Chip.IT8705F: 
         case Chip.IT8728F:
+        case Chip.IT8731F:
+        case Chip.IT8732F:
+        case Chip.IT8733F:
         case Chip.IT8771E:
         case Chip.IT8772E:
           gpioCount = 0;
diff --git a/Hardware/LPC/LPCIO.cs b/Hardware/LPC/LPCIO.cs
index 8fda66a..08a09d4 100644
--- a/Hardware/LPC/LPCIO.cs
+++ b/Hardware/LPC/LPCIO.cs
@@ -341,6 +341,9 @@ namespace OpenHardwareMonitor.Hardware.LPC {
         case 0x8721: chip = Chip.IT8721F; break;
         case 0x8726: chip = Chip.IT8726F; break;
         case 0x8728: chip = Chip.IT8728F; break;
+        case 0x8731: chip = Chip.IT8731F; break;
+        case 0x8732: chip = Chip.IT8732F; break;
+        case 0x8733: chip = Chip.IT8733F; break;
         case 0x8771: chip = Chip.IT8771E; break;
         case 0x8772: chip = Chip.IT8772E; break;
         default: chip = Chip.Unknown; break;

```

tds4u, 2017-03-21:
I found it - simple ^^
Enable "Show Hidden Sensors" via main menu to see all sensors.

closed 2017-03-21 by tds4u