openhardwaremonitor/openhardwaremonitor #957: AMD Ryzen Support for reading CPU voltage and Temperature

Scaredycrow, 2017-03-24:
Currently can read only load and core load metrics from Ryzen CPU's. 

Clocks, Voltage, Temperature and Fan speeds all appear to be missing (possibly also a need to support the new chipsets (X370, B350 etc)

bbaydar, 2017-03-24:
Seconding this request.

liabilityman, 2017-03-25:
Thirding

ooheyC, 2017-03-28:
Fourthing

Extarys, 2017-03-28:
It would be cool since no other linux hardware monitoring tool are still in dev sadly. +1

DarthDestroyer, 2017-03-28:
#961

No one, without a Ryzen,  could help you when nobody posts a report! 

The best known Answer to anything should be:
No help possible without a report :-/

Also, please see that you miss two (or more) parts! 
not only the cpu may be not supported, yet. 
the Mainboard is missing, too! 

Kind Regards

Soyasi, 2017-03-28:
hello!! look at this link https://github.com/groeck/it87/issues/10

cybercatgurrl, 2017-03-29:
Here's a hardware report. [OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/877564/OpenHardwareMonitor.Report.txt)

ThePunisher69, 2017-03-29:
Is there an echo in here?

https://github.com/openhardwaremonitor/openhardwaremonitor/issues/950

ooheyC, 2017-03-29:
The vendorID is 0x1463. But DeviceIoCntrol return error is 31. why?

DarthDestroyer, 2017-03-29:
@ooheyC, if you are familar with coding... there is a pull request to enumerate more informations for unknown cpus... maybe this could help? 

atleast, what i had seen from the less reports:
there seems to be an error at Estimated Time Stamp Counter Frequency Error: 0.51196 Mhz (source @cybercatgurrl)

liabilityman, 2017-04-08:
Here's another hardware report (Ryzen R7 1700). [OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/907129/OpenHardwareMonitor.Report.txt)

liabilityman, 2017-04-08:
@DarthDestroyer I'm not at all proficient with C#, but I think I managed to compile a version with @rmattuschka 's edits and run a report on my Ryzen system. Hope this helps! [OpenHardwareMonitor.Report.rmattuschka.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/907181/OpenHardwareMonitor.Report.rmattuschka.txt)

DarthDestroyer, 2017-04-08:
#836 is that, what i mean. 
i cant find, what there is added in your report...

liabilityman, 2017-04-11:
@DarthDestroyer Yeah I also tried #836 but I'm not sure what it's supposed to add either. 

Is there a debug version or debug mode I can run that would give you more information?

DarthDestroyer, 2017-04-11:
All Data that could be used for debugging is found in the report! you get all collected informations there, also errors or problems! 

that addition should show up in it! ;-)

Kraust, 2017-04-21:
@DarthDestroyer I Just submitted a report for my 1700. I would absolutely l-o-v-e Ryzen support so I can properly monitor temps and not need to rely on Ryzen Master.

DarthDestroyer, 2017-04-21:
that wasnt a failure, but i have to say that i cant access those ;-)
could you also upload your report to this issue?

Kraust, 2017-04-21:
[OpenHardwareMonitor-Kraust.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/947842/OpenHardwareMonitor-Kraust.Report.txt)

My bad, Have a report

Kraust, 2017-04-23:
In the mean time I am working on implementing support for the AX370 Gaming K7:
http://i.imgur.com/qEFUFPW.png

laffer1, 2017-05-12:
Here's another report for a Ryzen 1700 with the Asus Prime X370-pro
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/997766/OpenHardwareMonitor.Report.txt)

Extarys, 2017-05-17:
Is it possible to install this and generate a report under Linux?

NoXPhasma, 2017-06-11:
Under Linux it doesn't show anything beside the Processor and Motherboard name.

Board: ASUS Prime x370-Pro
CPU: AMD Ryzen 7 1700

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1066370/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2017-06-11:
It will not show more in linux! 

If you search "Linux" , it is well commented why and on what it depends :-(

NoXPhasma, 2017-06-11:
I don't understand, where should I search for Linux to find those dependencies? And just for the record, I can view lots of data with LMSensors (Voltages, Temperatures, Fan Speeds), I don't see why this shouldn't be possible with this software.

DarthDestroyer, 2017-06-11:
Go back to Issues (List), type into the Searchbox: "Linux" and you will get the posts about it! 

I didnt say, that it is impossible... Afaik, there is no one working on Linux Support, atm!

Kraust, 2017-06-11:
OT: I was looking into this issue again this weekend, and I still haven't found the PCI Device/Register where temperatures are located. It has most certainly changed and AMD hasn't released their BKDG for Family 17h yet. 

I did manage to get Bus Speed, as that register was available in a guide that they released last month.

On the Linux side I don't know of anyone who has implemented this either. If I could find out the PCI Device/Register where this data is held I could probably implement this fairly easily (Probably not on Linux tho, as I don't regularly use it).

DarthDestroyer, 2017-06-11:
For Linux,  this Issue was one of the Search-Results i had meant: #867

The MSR Addition from hirschmann looks not so bad, but i dont have a Linux with GUI available right now :-(

BennehBoy, 2017-07-13:
I found a fork of kraust's work with a branch that covers the ITE IT8665 - this gives another piece of the puzzle but is still WIP -> https://github.com/azuisleet/openhardwaremonitor/tree/IT8665E

This gives voltages, temps, and fans.  CPU temp I think needs some conditional code adding somewhere based on cpu type because it appears to include the offset (or part of!) - not sure it's possible to bake this into the superio area however and we may need to wait for the BKDG in order to update the AMD10 code.

JernejL, 2017-08-18:
Check https://linuxconfig.org/monitor-amd-ryzen-temperatures-in-linux-with-latest-kernel-modules - that apparently works.

Terrails, 2017-08-31:
Another report with Ryzen 5 1600 and B350 motherboard
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1267492/OpenHardwareMonitor.Report.txt)

Robyer, 2017-09-05:
Now it seems they finished it in https://github.com/groeck/lm-sensors/issues/16 (as @Soyasi posted this link before).

KyodaiKen, 2017-09-30:
Hey! I would like to send a report file for the Ryzen Threadripper 1950X on an ASUS ROG Zenith Extreme x399 chipset motherboard, but OpenHardwareMonitor then hangs and I can't even kill it, nor can I reboot. CPU-Usage is absolutely ZERO. I cannot help at all and would like to see all information available on my setup....

KyodaiKen, 2017-09-30:
Hello again. Here is a CPUz report instead attached.
[FELINE.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1346448/FELINE.txt)

Vahalu, 2017-10-27:
Mobo: X370 GAMING PRO CARBON
CPU: R5 1600x
Report: [OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1421045/OpenHardwareMonitor.Report.txt)

jumar, 2017-11-13:
isn't this issue closed by https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/17 ?

Kraust, 2017-11-13:
A pull request was never made to notify people that these changes were made. It seems like the person who made the fork doesn't intend to merge his changes back into this project (regardless of whether or not this repo is actually active or not).

Trip2009, 2017-12-07:
Another one for ASUS Prime B350-Plus
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1539217/OpenHardwareMonitor.Report.txt)

jctmf, 2017-12-13:
Another one for DELL INSPIRON 5675. X370 chipset
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1554079/OpenHardwareMonitor.Report.txt)

FireFrenzy, 2017-12-16:
HWInfo can read core temps, so its atelast POSSIBLE todo it and on my 1950x it seems to have the same 27 degree ofset everyone was freaking out about during the r7 launch... 

If i need to provide something more complex then screenshots for win10 64 Pro, let me know what to do and i'll hook you up...

dasMenz, 2018-03-27:
One more for ASUS CROSSHAIR VI HERO
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1853423/OpenHardwareMonitor.Report.txt)

FoxyRawrs, 2018-04-29:
Asus ROG Strix x370-f gaming 
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1958320/OpenHardwareMonitor.Report.txt)

Kallestofeles, 2018-07-14:
Mobo: ASUS Strix x470-f
CPU: Ryzen gen2 2700X
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/2195017/OpenHardwareMonitor.Report.txt)
Let's make the best monitoring tool even better. :)

DarthDestroyer, 2018-07-14:
@Kallestofeles

Chip ID: Unknown ITE with ID 0x8665 at 0x2E/0x2F

And you are using a outdated Version ;-)

Kallestofeles, 2018-07-14:
@DarthDestroyer 
The version I'm using is 0.8.0 beta which is the latest on the website: https://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0-beta.zip
If that's not the latest, then someone should update the webpage. :)
But I can confirm that all the data is there when using librehardwaremonitor branch build.

Kraust, 2018-07-14:
OHM is no longer supported and none of us have access to actually pushing new code changes. I would suggest you use HWiNFO as its really the only modern reliable hardware monitor out there.

DarthDestroyer, 2018-07-14:
http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

It isnt that bad and far from a Problem to get further with OHM! 
Fork it, Join the PRs and you have what you want! 

For only "using" there are enough options! But for me, it is the part of implementing into other tools why i try to help where i can! 
The Libre Guys are one Option, too!

mittyredstar, 2020-01-31:
ryzen 7 3700x with asus x470 prime pro no temps :((
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4139401/OpenHardwareMonitor.Report.txt)

alexandru2882, 2020-03-23:
Ryzen 7 2700X + MSI B450 TOMAHAWK
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/4370286/OpenHardwareMonitor.Report.txt)

moellerm, 2020-05-12:
This should be fixed in OHM version 0.9.0 and later.

closed 2020-05-12 by moellerm