openhardwaremonitor/openhardwaremonitor #959: Is this project still maintained in some way?

rmattuschka, 2017-03-25:


DarthDestroyer, 2017-03-25:
Of course!

Look at https://github.com/openhardwaremonitor/openhardwaremonitor/network!

If you are you are complaining about the acceptation of pull request, @moellerm is the project coordinator and is doing his best! Yeah, he isnt often present visible BUT, think about some Points:

1. its not his fulltime job
2. its open source - so, feel free to help yourself / all others
3. while it is open source, he has to check every pull-request / change
4. if you get not paid for your work, it isnt a must - what i want to say is, that it is a hobby!

Maybe now, you understand the situation better!
Feel free to extend the Code, help us to get all bugs out, do your best to implement new features...
I bet, @moellerm will then accept all good Pulls!

And please, dont waste the time (yes, every issue needs time - atleast for reading)...

rmattuschka, 2017-03-29:
@DarthDestroyer 

Who are you?
Is that your project?
You are not the sherrif!!

closed 2017-03-29 by rmattuschka

DarthDestroyer, 2017-03-29:
I dont see me as a sherrif, so be kind! 

Atleast me and some other are trying to maintain the project and we are collecting the needed informations and trying to help, where we could! 

Feel free to help and assist, everyone is invited! 

Keep calm! Kind Regards