openhardwaremonitor/openhardwaremonitor #964: No HTTPS/TLS for openhardwaremonitor.org

C0rn3j, 2017-04-04:
Hi, why does the site allow HTTP and the cert name the cert the HTTPS connection presents is wrong?

I can see that the cert name is for *.metanet.ch, and I'm assuming that's where the webhosting is.

For the sysadmin - you can use https://letsencrypt.org/ to get a free certificate and automate the renewal with certbot with a cron job.

Thermi, 2017-05-26:
I think the website has to support HTTPS, too, because it provides the binaries to users. Without authenticated TLS, it is too easy to inject malicious code with a MITM attack.

DarthDestroyer, 2017-05-26:
You arent wrong, but that is the smallest problem at this repo ;-)

Ajedi32, 2017-12-11:
Yeah, as cool as this project is, I'm not comfortable downloading and running unsigned binaries over an unencrypted connection. Couldn't you at least attach the final builds to the GitHub tags so I can download them from GitHub directly?

DarthDestroyer, 2017-12-11:
Best thing about open source: you can do anything by yourself ;-)
You dont have to download it from the ohm page, you dont have to run unsigned binaries! 
Download the Source, compile and sign it yourself and you are done ;-)

But, you arent really wrong... let us be patient

sunk818, 2017-12-11:
>  I'm not comfortable downloading and running unsigned binaries over an unencrypted connection

If you're that uncomfortable, maybe you should not use the software. Paranoid much?