openhardwaremonitor/openhardwaremonitor #965: Asus TUF Z270 MARK 2

antonyb2, 2017-04-05:
No info.. Help!

DarthDestroyer, 2017-04-05:
search for "z270" and read, there is help! 

or, because im nice today: #926 :-P

antonyb2, 2017-04-05:
Thank's..

antonyb2, 2017-04-05:
CPU now works with Alpha version, missing fans, temperature ssd ..

DarthDestroyer, 2017-04-05:
like the other issues: without a report, its impossible to help you further! 

and, i would say, that you append to the other issue for that - because it is the same issue ;-)

antonyb2, 2017-04-05:
Thank's..

[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/900463/OpenHardwareMonitor.Report.txt)
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/900464/OpenHardwareMonitor.Report.txt)

antonyb2, 2017-04-26:
There are updates?