openhardwaremonitor/openhardwaremonitor #968: Fan starts and stops repeatedly

Gettyk, 2017-04-20:
When Open Harware Monitor is running, my fan starts spinning and after 2 or 3 seconds stops then it starts again an then stop. 

I tried the 0.8.0 beta version as well as the 0.7.1 Beta and 0.6.0 Beta. I also tried [SidebarDiagnostic](https://github.com/ArcadeRenegade/SidebarDiagnostics/releases/tag/3.4.5) which is based on  Open Hardware Monitor.

My computer is a Dell Xps17, running with Windows 7,  Intel(R) core(TM) i7-2760QM CPU 2.40GHz  2.40 GHz. 

The fan is acting weird even when I disabled my firewall (comodo). 

ON my second computer, a Dell Inspiron 3520 Intel(R) Core(TM) i3-2370M CPU @ 2.40GHz (4 CPUs), ~2.4GHz, I have no such issue.

DarthDestroyer, 2017-04-20:
Yeah, some Laptops will not work! 

There was a similar Issue before, that could be found... 
It was something with related to Fan Controller...

DarthDestroyer, 2017-05-02:
I think i found it! => #868

Gettyk, 2019-05-04:
I updated my computer to Windows 10 built 1809 and OHM works fine now.