openhardwaremonitor/openhardwaremonitor #969: Big discrepancy between openhardwaremonitor CPU total load and Windows task manager

BlueNinja0, 2017-04-20:
I'm on OpenHardwareMonitor 0.8.0 (although I also observed this discrepancy in 0.7.0) on Windows 10 Pro x64.

When taxing my CPU (Intel Core i5-3570K @ 4.5Ghz) with a relatively stable workload, like running a game in a static situation, I get very different CPU load totals on OpenHardwareMonitor versus what Windows Task Manager reports. OpenHardwareMonitor always reports much lower values, for example, if Windows shows 80% CPU usage, OpenHardwareMonitor might show 50 or 60%.

DarthDestroyer, 2017-04-21:
that seems possible because your cpu is overclocked! 
Windows is a little bit to stupid to get the real core clock, that will result in a lower core clock than it really is and the math of the usage creates a higher percentage! 

OHM collects the real data more ore less directly out of the cpu itself, with the real core clock! puts that in math and shows you the correct usage. 

the win Version is nearly ignorable, because they do it all ;-)

it isnt tested, how much diff that makes! You could try it out yourself ;-)

BlueNinja0, 2017-04-21:
But Windows appears to get the correct core frequency in real time:

[image]

It's still a possibility though. I'll try to get around and test it with the CPU's stock clock.

DarthDestroyer, 2017-04-21:
Yep, correct! 

But, look at "Maximum Speed" ;-)

And now, when you know the math to get percentage, you will see the problem ;-)

DarthDestroyer, 2017-04-23:
What did your Research said? Am i right with my Theory?

BlueNinja0, 2017-04-28:
I just did a few tests at various frequencies and indeed you seem to be correct! The closer I test to the stock frequency (3.4Ghz in this case), the closer the % between OpenHardwareMonitor and the Task Manager.

At 4.5Ghz, the Task Manager shows load about 20% higher than OHM when testing a stable load.
The same test at 3.4Ghz (the stock frequency), the Task Manager shows load only 2-3% higher than OHM, which I think we can assume to be some rounding errors.

It's a shame Windows can't even get the CPU load right...

DarthDestroyer, 2017-04-28:
yeah, correct ;-) but, when Windows would work, ohm would be useless :-D

nice to know, that you can prove it! thanks for your effort!

BlueNinja0, 2017-04-28:
Definitely not worthless! It has so many awesome features. CPU load is just one of them.

closed 2018-05-11 by BlueNinja0