openhardwaremonitor/openhardwaremonitor #972: GPU Core Voltage Erroneous

Kronix420, 2017-04-30:
[image]

Just got this card today. It's the rx 580 8gb
Drivers are all installed, and I used DDU prior to installing the card. I have done multiple restarts after finalizing the amd drivers, so I don't think that's the issue.

DarthDestroyer, 2017-05-02:
I cant help you, at that point... But you are not alone!
Found some other Issues similar to yours. => #825

Could you set a better title for your Issues? Something like "Radeon RX 580 wrong Core Voltage"?
It seems happening only with AMD/ATI Cards!

Maybe, it would help to get a Report from you, too!

Kronix420, 2017-05-02:
I think figured it out. My card is the gigabyte rx 580 8gb, and gigabyte does not allow editing of the fan or clocks from anything except their Aorus software. I'm assuming they have implemented a system in which the user can only get accurate results from their software. I was hoping that you just had to add the 580s to the program since they're new but if not it's most likely this brand I think

DarthDestroyer, 2017-05-02:
I would never say, that this would be impossible... but it would be crazy if they would edit the ati/amd driver at that aspect only to block other applications!

like the other issue, there is a problem where the value gets wrong... 
A Report would really help here! 
I dont think that its not solvable...

myfreeweb, 2017-07-24:
Same problem happens with the reference RX 480, this has nothing to do with the board vendor.

I'll look into it…

DarthDestroyer, 2017-07-24:
that would be good because i'm out at that problem! /me is Nvidia User

rashom, 2017-07-28:
I have 2 PCs with Radeon cards (RX 480 & RX 560) both showing erroneous ultra high voltages.
My third pc has the FirePro W7000, will check later.

myfreeweb, 2017-07-28:
So the Overdrive API just returns a weird value, that happens in the API samples too.

The newer Overdrive versions don't seem to do any better — `ADL2_OverdriveN_PerformanceStatus_Get` returns a VDDC of zero:
[image]

HWiNFO [had that problem but fixed it *somehow*](https://www.hwinfo.com/forum/Thread-Custom-Polaris-GPUs-RX-480-incorrect-parameters). Not by talking to the voltage regulator because that stuff is a separate section:

[image]