openhardwaremonitor/openhardwaremonitor #973: openhardwaremonitor development

lucascosti, 2017-05-01:
While a lot of us patiently wait for updates to this wonderful program to support recent chipsets (e.g z270 boards, Ryzen, etc.), perhaps it might be useful to discuss if the current maintainer (@moellerm) is doing enough to keep up with changes, and responding to well-meaning PRs. One update in ~2.5 years is quite disappointing.

I'm no developer, but I am happy to lend my expertise as a tech writer in any way I can. Maybe it is time for a proper fork to continue development? If anyone has any dev expertise, let us know. @DarthDestroyer seems to be great in helping people and responding to issues; @DarthDestroyer do you know anyone?

DarthDestroyer, 2017-05-01:
thanks for the nice words! 
i try to help, where i can, and that nearly everytime or let me say: when my mobil phone pushes a notification :-D

so, btt! 
The Problem is not @moellerm himself... yeah, he is the head of ohm and that is not a small "job", you have to understand, that it isnt a mainjob and he gets no money out of it. I dont know, what trouble or problems are in his live and i would never judge about that... 
But, i have to say, that he should atleast add a second admin (or someone who can acceppt pulls). 
Im a Coder myself and i love that little Tool! This is the reason why i try to help!

@moellerm was here around the new year, if i remember correctly! 

To your Fork Idea: i had tried that, and it worked atleast on another way - my fork got forked :-D 
That was crazy to add the new and last additions out of the Original... 
But  it is the only thing, we can do without @moellerm... :-/

ragesaq, 2017-05-02:
@DarthDestroyer I have read a lot of your contributions to both issues and commits. It would be pretty great if @moellerm were to make him an admin to keep this project moving!

martynsmall, 2017-05-05:
Agreed, I'm seeing a lot of updates for additional support which people are adding that simply aren't available to those not familiar with development.  Someone else that can accept pulls would be great!

GoldMath, 2017-05-23:
I support this too.
One year to two for a update is way too much.

markau, 2017-07-13:
Such a shame to let a popular and useful open source project wither on the vine.

DarthDestroyer, 2017-07-13:
It wouldnt be so hard, if we could get us together to maintain a fork, name it "Experimental" and if we get solutions / additions, pull them to main...

BennehBoy, 2017-07-14:
There are quite a few ITE SuperIO implementations that should be straightforward to implement (thanks to the it87 linux module) - I have IT8603 preliminary support added for an A88XM based motherboard that I own.  Happy to submit a PR but it's of more use if it can make it into an active release.