openhardwaremonitor/openhardwaremonitor #975: OHM not finding any readings on Ubuntu 16.04

nkotsianas, 2017-05-04:
Hey all, I've seen other peoples' success with OHM reading some temp and voltage values, but my app simply shows no readings.

Mono seems to be working fine. I was able to replicate and run the "Hello World" apps on the [Mono site](http://www.mono-project.com/docs/getting-started/mono-basics/) after installation.

Issue report: [OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/975260/OpenHardwareMonitor.Report.txt)
Screenshot (showing Psensor values for reference): [image]

DarthDestroyer, 2017-05-04:
#835 is a similar Issue!

nkotsianas, 2017-05-04:
@DarthDestroyer thanks for your response (and your dedication to this project)!
Yeah I saw that issue, but at least they have some readings, where I have nothing at all. Is it a problem with identifying the CPU/temp sensors on my mobo/system?
So I did as you suggested in that thread: I'm using the latest version (0.8.0-beta), and I posted the Report file. Do you know if there is anything else I can try?

DarthDestroyer, 2017-05-04:
The other Issue is a little bit old, so the "last" Version is http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

But that will not help you, at that Point!

To your Point:
There are several reasons why this might not work!
- Missing permissions (tried it with sudo?)
- Mono
- Linux itself

What i can say for now: Mainboard will not work and Nvidia GPU, too!

Is it a Dual-Boot System, a Windows Report would give correct missing Hardware-Support!

nkotsianas, 2017-05-04:
Thanks again for your suggestions! Running with sudo did not help unfortunately! :(

I do have a dual boot system with Win10 on a separate physical drive. Here's the Report file from that installation: [OpenHardwareMonitor.Report-Win10.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/977263/OpenHardwareMonitor.Report-Win10.txt)

Everything seems to run great on the Windows 10 side. Does this indicate that the problem can be fixed on my machine?

DarthDestroyer, 2017-05-04:
The Windows Report helps a little bit!

For the Mainboard: Missing Compatibility (only for further investigation!)
Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0x8587 at 0x2E/0x2F
Chip ID: Unknown ITE with ID 0x8587 at 0x2E/0x2F
Chip ID: Unknown SMSC with ID 0x8587 at 0x2E/0x2F

Im sorry, but im not into the Linux Part - so far... I dont know, how it exactly work and how much is depending on Mono and its functions!
I think, that Mono is the bottleneck at that Point =/

Maybe, the Linux-Guys could help here?

nkotsianas, 2017-05-04:
Okay, so it's probably something to do with Linux/Mono. I will see if a someone who knows more Linux can comment. Thank you for your help!

hirschmann, 2017-05-05:
The problem is neither Mono, nor Linux itself. OHM's Linux support is just very limited.
This is the same issue as #867

DarthDestroyer, 2017-05-05:
That was the Issue i had searched -.- 
Merci!

nkotsianas, 2017-05-06:
@hirschmann thanks for the info! I guess we can only wait and hope to get more support for Linux in the future! Much love to this project though, the community appreciates the effort!