openhardwaremonitor/openhardwaremonitor #978: Mainboard ASUS STRIX Z270H GAMING can't get info and I7-7700 can't get temperatures

Vihe91, 2017-05-11:
[image]

My OpenHardwareMonitorLib version is 0.80
Request repair, thank you!!!

Vihe91, 2017-05-11:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/992596/OpenHardwareMonitor.Report.txt)

Vihe91, 2017-05-11:
[OpenHardwareMonitor.Report.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/992597/OpenHardwareMonitor.Report.txt)

DarthDestroyer, 2017-05-11:
http://openhardwaremonitor.org/files/openhardwaremonitor-v0.8.0.3-alpha.zip

Your Mainboard Support is pending! See #884
(feel free to implement it by yourself) 

Regards

Vihe91, 2017-05-12:
Already normal, thanks for your reply!!!

brunis, 2017-05-18:
I had the same issue with kaby lake i5-6700k, no temps or clockspeeds. v8.0.3 alpha solves all, thanks!