openhardwaremonitor/openhardwaremonitor #979: permanent winring registration

Oozza, 2017-05-12:
Hello, 
is it possible to register winring driver permanently so it hasn't to be registered with every use of OpenHardwareMonitorLib.dll?  I am trying to use this library in a command-line tool that is periodically called by Zabbix Agent for purposes of remote monitoring.

DarthDestroyer, 2017-05-12:
Maybe is that possible, but not practicable at ohm level... 
I cant say, how i would help there because i dont know Zabbix?! 
If you can feed that, with informations, i would go another way (or 3)! 

1. let your command line tool running while a command for refresh will echo new values
 
2. get the data from the webserver included in ohm

3. get the data from wmi while ohm is running

I use the lib also on a project, but i do it like ohm... 
for deeper investigation, i would create a api wrapper, if i were you ;-)

Oozza, 2017-05-12:
Zabbix is a web server used for remote monitoring of hosts. It usually periodically queries hosts for metrics such as CPU utilization or network availability. There are plenty of ways how to get various data from these hosts, including WMI or queries on web services.

Command-line script using OHM lib to get one sensor value at a time was easy to implement and straightforward solution, but later I noticed many service registration events in System's event log.

For now, I will try the WMI way (3) and consult it with the team. 

If that doesn't work I will try to find out what OHM web server does.

Thanks for your response.

ragesaq, 2017-05-12:
@Oozza maybe check this out for inspiration: https://github.com/vigne/ohw2carbon
A python app that pulls the OpenHardware WMI data and shoves it into graphite

hirschmann, 2017-05-12:
@Oozza
My fork of OpenHardwareMonitorLib installs the driver permanently: https://github.com/hirschmann/nbfc/tree/master/Core/Plugins/OpenHardwareMonitor

closed 2022-03-07 by Oozza