openhardwaremonitor/openhardwaremonitor #980: Show Plot(0.8.0 Beta) not draw anything on Microsoft Windows [Version 10.0.15063, X64, I5]

alanackart, 2017-05-21:
As the title has said, Show Plot show only this:

[image]

martynsmall, 2017-05-21:
Checkboxes appear next to each item in the main window when you open the plot.  These let you select which items to show, and without any ticked it will appear blank.

alanackart, 2017-05-22:
thank for your replay, I have followed your instructions and indeed it works.
[image]

 I am sorry to bring you extra burden and I would be more careful to open an issue in Github.

closed 2017-05-22 by alanackart