openhardwaremonitor/openhardwaremonitor #981: ASUS STRIX Z270F GAMING MAINBOARD Support Request

Mindvampire, 2017-05-30:
CPU and CHASSIS FAN Speed missing

Mindvampire, 2017-05-30:
Open Hardware Monitor Report

--------------------------------------------------------------------------------

Version: 0.8.0.3

--------------------------------------------------------------------------------

Common Language Runtime: 4.0.30319.42000
Operating System: Microsoft Windows NT 10.0.15063.0
Process Type: 64-Bit

--------------------------------------------------------------------------------

Sensors

|
+- ASUS STRIX Z270F GAMING (/mainboard)
|
+- Intel Core i7-7700K (/intelcpu/0)
|  +- Bus Speed      :      100  99.9935  100.007 (/intelcpu/0/clock/0)
|  +- CPU Core #1    :  4500.01  799.973   4500.3 (/intelcpu/0/clock/1)
|  +- CPU Core #2    :     3300  799.973   4500.3 (/intelcpu/0/clock/2)
|  +- CPU Core #3    :     2400      800   4500.3 (/intelcpu/0/clock/3)
|  +- CPU Core #4    :  4500.01      800   4500.3 (/intelcpu/0/clock/4)
|  +- CPU Core #1    :       32       31       58 (/intelcpu/0/temperature/0)
|  +- CPU Core #2    :       36       33       64 (/intelcpu/0/temperature/1)
|  +- CPU Core #3    :       33       31       63 (/intelcpu/0/temperature/2)
|  +- CPU Core #4    :       33       31       63 (/intelcpu/0/temperature/3)
|  +- CPU Temperature :       36       33       64 (/intelcpu/0/temperature/4)
|  +- CPU Total      :  1.95313 0.585938  31.6406 (/intelcpu/0/load/0)
|  +- CPU Core 1     :        0        0       50 (/intelcpu/0/load/1)
|  +- CPU Core 2     :   1.5625        0  48.4375 (/intelcpu/0/load/2)
|  +- CPU Core 3     :   4.6875        0  41.4063 (/intelcpu/0/load/3)
|  +- CPU Core 4     :   1.5625        0     37.5 (/intelcpu/0/load/4)
|  +- CPU Power Consumption :  15.3109  8.00619  46.5739 (/intelcpu/0/power/0)
|  +- CPU Cores      :  4.15654  1.51972  35.4014 (/intelcpu/0/power/1)
|  +- CPU Graphics   :        0        0        0 (/intelcpu/0/power/2)
|  +- CPU DRAM       : 0.929406 0.916829   1.7393 (/intelcpu/0/power/3)
|
+- G-Skill Trident X (/ram)
|  +- Memory         :  24.7875  22.6612   25.295 (/ram/load/0)
|  +- Memory Used    :  7.91518  7.23621  8.07723 (/ram/data/0)
|  +- Memory Available :   24.017  23.8549  24.6959 (/ram/data/1)
|
+- NVIDIA GeForce GTX 970 (/nvidiagpu/0)
|  +- GPU Core       :      135      135  1113.75 (/nvidiagpu/0/clock/0)
|  +- GPU Memory     :      324      324   3505.5 (/nvidiagpu/0/clock/1)
|  +- GPU Shader     :      270      270   2227.5 (/nvidiagpu/0/clock/2)
|  +- GPU Temperature :       40       39       44 (/nvidiagpu/0/temperature/0)
|  +- GPU Core       :        1        0       49 (/nvidiagpu/0/load/0)
|  +- GPU Memory Controller :        7        1       32 (/nvidiagpu/0/load/1)
|  +- GPU Video Engine :        0        0       24 (/nvidiagpu/0/load/2)
|  +- GPU Memory     :  9.80711  8.71325   15.898 (/nvidiagpu/0/load/3)
|  +- GPU Fan Speed  :        0        0        0 (/nvidiagpu/0/fan/0)
|  +- GPU Fan        :        0        0        0 (/nvidiagpu/0/control/0)
|  +- GPU Memory Free :   3694.3  3444.82  3739.11 (/nvidiagpu/0/smalldata/1)
|  +- GPU Memory Used :  401.699  356.895  651.184 (/nvidiagpu/0/smalldata/2)
|  +- GPU Memory Total :     4096     4096     4096 (/nvidiagpu/0/smalldata/3)
|
+- WDC WD20EARX-00PASB0 (/hdd/0)
|  +- HDD Temperature :       34       34       35 (/hdd/0/temperature/0)
|  +- HDD            :  65.2193  65.2166  65.2193 (/hdd/0/load/0)

--------------------------------------------------------------------------------

Parameters

|
+- ASUS STRIX Z270F GAMING (/mainboard)
[... 737 lines cut ...]
 04 Start/Stop Count                   500200000000 100   100   0     592     
 05 Reallocated Sectors Count          000000000000 200   200   140   -       
 07 Seek Error Rate                    000000000000 200   200   0     -       
 09 Power-On Hours (POH)               589C00000000 46    46    0     40024   
 0A Spin Retry Count                   000000000000 100   100   0     -       
 0B Recalibration Retries              000000000000 100   100   0     -       
 0C Power Cycle Count                  490200000000 100   100   0     585     
 C0 Emergency Retract Cycle Count      4B0000000000 200   200   0     -       
 C1 Load Cycle Count                   EC1105000000 90    90    0     -       
 C2 Temperature                        220000000000 105   116   0     34      
 C4 Reallocation Event Count           000000000000 200   200   0     -       
 C5 Current Pending Sector Count       010000000000 200   200   0     -       
 C6 Uncorrectable Sector Count         000000000000 200   200   0     -       
 C7 UltraDMA CRC Error Count           000000000000 200   200   0     -       
 C8 Write Error Rate                   000000000000 200   200   0     -       

Logical drive name: C:\
Format: NTFS
Total size: 800156807168
Total free space: 360828256256

Logical drive name: D:\
Format: NTFS
Total size: 1180184428544
Total free space: 327949025280

DarthDestroyer, 2017-05-30:
Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xD121 at 0x2E/0x2F

DarthDestroyer, 2017-05-30:
Why do you not add to the Issue 2 posts down?!  #978

There is already a homemade Solution! 
The Pull Request is already opened and is waiting to be approved! 

For your RAM Issue, i hadnt seen any RAM Temps in OHM. I will try to look at, when i got home... or tomorrow

Mindvampire, 2017-05-30:
Hi Darth,

as you can see in my profile I am already using the altered version 0.8.0.3 but the CPU FAN and RAM TEMP is still missing.

Thanks for considering.

DarthDestroyer, 2017-05-30:
You are correct, but that is not the Problem ;-)

After the Link is the Pull Request #884 ! That is the Solution ;-)

Mindvampire, 2017-05-30:
Ok call me retard in this regard but I clicked your link and there is no download.
What do I have to do to make it work?

Thanks

DarthDestroyer, 2017-05-30:
There are many ways to rome ;-)

A Pull Request is an addition to the Sourcecode! That is viewable at the the #884. 

You could fork OHM and accept that pull request and download the Sourcecode... or you could download the ohm Sourcecode and change it per hand. 

For everything is a small or little knowledge of coding needed... there is nothing "ready to run available", yet :-/

Mindvampire, 2017-05-30:
Ah ok, I guessed it would be this way. 

I am able to code it but I have no time to setup a build environment right now, so I hoped for an already compiled download.

Thanks for the info.

DarthDestroyer, 2017-05-30:
Anyone could provide the compiled version... but do you trust us? 

So, get Visual Studio Community / Express and double click the mainfile in the Source... Done ;-)

ragesaq, 2017-05-31:
@DarthDestroyer 
Is there a simple guide you could put together so somebody with no VS experience could complete what you are proposing?

ragesaq, 2017-06-02:
@DarthDestroyer I figured it out. I installed VS 2017 community edition and added .net desktop applications, then I could build. Sadly this did not fix my issue #939 because the NCT6791D is the chip in question. My further comments will go there.

DarthDestroyer, 2017-06-02:
nice! congrats! 

then, when you can build, take a look at the pull request #884! 
You can see the file, the Original content at that line (red) and the new or changed code (green). 
Do that, to your Sourcecode and build it => done ;-)

by the wa, set the mode not to debug! use release instead, if possible...

DarthDestroyer, 2017-06-02:
https://github.com/openhardwaremonitor/openhardwaremonitor/pull/884/files

that is the changelog

ragesaq, 2017-06-02:
lol, just because I figured out how to get VS to build doesn't make me a programmer, I have no idea what I'm doing with this project, I just wanted to try and get the rest of my sensors to show up. Anyhow, as mentioned previous, pertinent comments to my issue in the ticket I opened up #939

DarthDestroyer, 2017-06-05:
Let me write a quick Tutorial!

Open OHM in VS
At the right upper side, you should got a filebrowser like windows.
Open the Pull #884 - changes and search the listed files! With Ctrl+G you can go to the lines that got changed. Replace the Red-Lines with the Green-Lines. All of them in all Files that are listed there!
After that, you build the Project and should get a positive Result on your Sensors!

I know, that this doesnt make you a Pro-Programmer ;) But you did 80% of the work for your Solution - why quit before a Result?

For that, you dont have to think, what you are doing... If you can change a Word-Document, you can combine the Pull-Request with the Latest Source-Code! I bet!

ragesaq, 2017-06-05:
@DarthDestroyer 
That is what I figured out and have tried, however, as I mentioned both times previously, this did not resolve my issue as I do not have the same Asus board. As an attempt to not derail this particular ticket and its reflection of the issues regarding the Strix Z270F motherboard, I attachment my comments reflecting my situation are in the ticket I opened. The ticket in quesiton is #939 as previously mentioned.

DarthDestroyer, 2017-06-06:
Yes and No ;-)

I understand what you are searching and what the problem is... But there are two construction sites, right now! 

#884 should give you the NCT6793D Controller Results, it doesnt matter that your Mainboard Modell differs by a Letter or Number as far as long the Controller is correct! 
That could be double checked with HWiNFO aswell! Category Mainboard and there should be a Controller Thing, with an ID like NCT6793D. 
If that differs, feel free to say that here! 

Second Part, is you additional Controller, about that we talk in the other issue!

ragesaq, 2017-06-06:
My motherboard doesn't have an NCT6793D chip, it has an NCT6791D.

DarthDestroyer, 2017-06-06:
Correct, and it seems to be the same ;)

I can see, that it works - from the other Post... So, nevermind!