openhardwaremonitor/openhardwaremonitor #982: Can support HP ProLiant DL580 Gen8 @ Windows Server 2012 ?

syncyao, 2017-06-01:
"The free Open Hardware Monitor software runs on 32-bit and 64-bit Microsoft Windows XP / Vista / 7 / 8 / 8.1 / 10 and any x86 based Linux operating systems without installation."

But, I still test it on HP ProLiant DL580 Gen8 with Windows Server 2012.
And, there is no SMART info.

[OpenHardwareMonitor.Report 100.txt](https://github.com/openhardwaremonitor/openhardwaremonitor/files/1044340/OpenHardwareMonitor.Report.100.txt)

DarthDestroyer, 2017-06-01:
lol... and yes, that advertisement sentence is correct... or how do you know if it doesnt work? :-P

BTT: You have a Server! There are many differences in hardware and the accessing compared to desktops... for example the fan and temp Controller, Memory or maybe HDDs, also! 

For later Investigation:
Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0x401 at 0x2E/0x2F

Chip ID: Unknown ITE with ID 0x401 at 0x2E/0x2F

Chip ID: Unknown SMSC with ID 0x401 at 0x2E/0x2F

syncyao, 2017-06-02:
Thanks!

And more info:
I have tried AIDA64 Business ( 2 249.00 USD, 3 years, monitor 100 PCs ) yesterday ^o^,
And it can not identify my HP server's SMART too →_→.
I doubt that it really exist SMART.

I have a question, so sorry for my poor English, I come from China, does "For later Investigation" mean that, I need to shutdown my machine and check the board? Or that is Open Hardware Monitor's following work?

DarthDestroyer, 2017-06-02:
no Problem! im totally fine with your english, mine isnt better ;-) i am in germany... 

dont worry at that! that is for ohm, for example searching which doesnt look into txt files! 

btt: does the manufacturer delivers an own application for the hdds/smart?

syncyao, 2017-06-02:
~_~

Ah, that accounts for it!

But I don't think it's open source or free or exist.
Anyhow I'll take a look on manufacturer's web.

DarthDestroyer, 2017-06-02:
Maybe not, but it should tell you if Smart is supported or not! 

WD has its own tool and your manufacturer maybe, too! 
for ohm it isnt a really help

closed 2018-08-13 by syncyao