openhardwaremonitor/openhardwaremonitor #985: Provide signed releases

Abdull, 2017-06-07:
It becomes more and more dangerous to install software, cf. the [HandBrake incident](https://forum.handbrake.fr/viewtopic.php?f=33&t=36364). Having a signed installer / executable would be great. As an intermediate, maybe you can provide executables at the GitHub repository.

DarthDestroyer, 2017-06-07:
I am totally on your side with the point "Provide Releases through GitHub". That would be a great addition!

At the Risk / Having an unsigned Executable, i am not! As long as it is Open-Source, there is no really need for that! Anyone could download the Source and compile it on its own. There is no Installer with bloatware or malware here...
It should be the right way, to sign it... But there is no really need, imho!

Abdull, 2017-06-07:
I followed your suggestion and to build OpenHardwareMonitor from sources (`tags/0.8.0`).

8 gigabytes of *Visual Studio Community* and a VS-suggested installation of .NET Framework 2.0 later, I was able to get an executable distribution by building OpenHardwareMonitor **and** OpenHardwareMonitorLib, then copying the `Bin/Release/` folder.

My Visual Studio skills are nonexisting, so maybe it may help others as well to provide a small README which explains how to build OpenHardwareMonitor from sources?

DarthDestroyer, 2017-06-07:
Yeah, a Tutorial for Beginners would be a good Idea!

But as you see, it wasnt very hard... ;)