openhardwaremonitor/openhardwaremonitor #986: System tray icon present but OHM window does not open.

Jonkieber, 2017-06-07:
I have an issue upon opening OHM, the icon in the system tray is present but is not right-clickable. The main window that normally appears on program start does not open. When double clicked, the system tray icon does not open the main window. The only way I can get the main OHM window to appear is to open the program twice, the window will appear every time if there is already an OHM process in task manager. If there is no OHM process already open the main window will not appear. There is no error in Event Viewer corresponding to this.

This is occurring on a total of 17's computers all running windows 7 pro, OHM is running as admin on all machines. The machines have the following motherboards.

Asus P8Z77-V
Asus P8Z68-V
Asus P8Z68-V Gen 3
Intel DP55KG

Any assistance with this would be greatly appreciated.

DarthDestroyer, 2017-06-07:
What Version of OHM do you use?

Jonkieber, 2017-06-07:
I am experiencing this same behavior on 0.7.1.0 & 0.8.0.0

DarthDestroyer, 2017-06-11:
I had tried 7 own PCs... Can reproduce your Problem! 
All running Win10Pro

Jonkieber, 2017-06-12:
I am also unable to reproduce the issue on my Windows 10 machines. It seems isolated to Windows 7 Pro at the moment. I am hoping to find a way to debug this issue and resolve it but have not yet found a method to do so.